AlexandreRouma/SDRPlusPlus #1356: Full IQ on SDR++ Server

Streety66, 2024-03-12:
Connecting remotely to a RPi4 from windows desktop using SDR++ Server as the source and running sdrpp -server on the RPi - Full IQ option is ticked and I cannot untick it, to reduce bandwidth and increase performance.

Is this deliberate? Why have a tick box if you cannot untick it?

Many thanks Alexandre.

Paul.

AlexandreRouma, 2024-03-12:
> Why have a tick box if you cannot untick it?

Because the feature will be added later.

Streety66, 2024-03-12:
Thank you, Alexandre.

Paul.

zodoczi, 2026-10-03:
Its working now on my fork, for ESP32-S3 turbo mode 80Mhz FFT only visualization (no real IQ data as that cannot be streamed out from the tiny MCU (yet))

the ESP32-S3 that cannot handle wideband IQ data over USB but *can* provide on-chip calculated spectrum data up to 80 MHz. Currently, there is no way in SDR++ for a source to feed a pre-computed spectrum directly to the waterfall display.

Connection to #1356: this same missing capability is also needed for the "Server FFT-only" mode (to save bandwidth).
The question: do you plan to implement this, or would you accept a feature request allowing source modules to optionally provide pre-computed FFT data? 

I confess Claude helped too!

https://github.com/z2labs/sdrpp-esp-sdr-source/issues/1

Thanks,
Zoltan