AlexandreRouma/SDRPlusPlus #1360: Crash starting with hackrf on Linux Mint

peterbmarks, 2024-03-14:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel Core i5-6300 @ 2.4HGz x 2
- RAM: 7.6GB
- GPU: Intel Skylake GT2 [HD Graphics 520]
- SDR: Local HackRF

**Software**
Linux Mint 21.3 Cinnamon
SDR++ v1.1.0 Built at 21:36:19, Feb 24 2024

**Bug Description**
A clear description of the bug.

SDR++ crashes when a hackrf is connected and click the start button.

**Steps To Reproduce**
1. Connect hackrf
2. Run SDR++ and select the hackrf device, the serial number is shown correctly
3. Click the start button and the program exits in a few seconds.

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Linux: Open a terminal and run `sdrpp -c`
I tried -c but it isn't recognised.

marksp@latitude:~$ hackrf_info 
hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c82253235f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x0062435a
marksp@latitude:~$ sdrpp
[15/03/2024 08:02:51.000] [INFO] SDR++ v1.1.0
[15/03/2024 08:02:51.000] [INFO] Loading config
[15/03/2024 08:02:51.000] [INFO] Using OpenGL 3.0
[15/03/2024 08:02:51.000] [INFO] Loading icons
[15/03/2024 08:02:51.000] [INFO] Loading band plans
[15/03/2024 08:02:51.000] [INFO] Loading band plans color table
[15/03/2024 08:02:51.000] [INFO] Loading modules
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/iq_exporter.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/pager_decoder.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[15/03/2024 08:02:51.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/limesdr_source.so: libLimeSuite.so.20.10-1: cannot open shared object file: No such file or directory
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/recorder.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_client.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/radio.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/m17_decoder.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/perseus_source.so
[15/03/2024 08:02:51.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/perseus_source.so: libperseus-sdr.so.0: cannot open shared object file: No such file or directory
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[15/03/2024 08:02:51.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_source.so
[... 78 lines cut ...]
[15/03/2024 08:02:53.000] [INFO] Running post-init for HackRF Source
[15/03/2024 08:02:53.000] [INFO] Running post-init for Hermes Source
[15/03/2024 08:02:53.000] [INFO] Running post-init for Network Sink
[15/03/2024 08:02:53.000] [INFO] Running post-init for PlutoSDR Source
[15/03/2024 08:02:53.000] [INFO] Running post-init for RFspace Source
[15/03/2024 08:02:53.000] [INFO] Running post-init for RTL-SDR Source
[15/03/2024 08:02:53.000] [INFO] Running post-init for RTL-TCP Source
[15/03/2024 08:02:53.000] [INFO] Running post-init for Radio
[15/03/2024 08:02:53.000] [INFO] Running post-init for Recorder
[15/03/2024 08:02:53.000] [INFO] Running post-init for Rigctl Server
[15/03/2024 08:02:53.000] [INFO] Running post-init for SDR++ Server Source
[15/03/2024 08:02:53.000] [INFO] Running post-init for SDRplay Source
[15/03/2024 08:02:53.000] [INFO] Running post-init for SoapySDR Source
[15/03/2024 08:02:53.000] [INFO] Running post-init for SpyServer Source
[15/03/2024 08:02:53.000] [INFO] Ready.
Segmentation fault (core dumped)
marksp@latitude:~$ 

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
I don't know when this started.

Thank you for a wonderful program!

AlexandreRouma, 2024-03-14:
duplicate of #337 #488 #760 #819 #1072 (and many others...). It's a conflict caused by libhackrf not checking if it's already inialized. There is **nothing** I can do about it.

The solution has been in the readme since almost day 1 of the project, please search existing issues and have a look at the documentation before opening an issue.
[image]

closed 2024-03-14 by AlexandreRouma

peterbmarks, 2024-03-14:
I'm very sorry to waste your time. I did search for any **open** bugs about HackRF but found none. Perhaps you should leave 1 open bug? (I realise it's not your bug).