AlexandreRouma/SDRPlusPlus #1366: RTL-SDR v4 white noise on Linux Raspberry Pi 5

wptracy, 2024-03-27:
I have the same problem as: 
"[No signal only white noise with RTLsdr v4 #1340](https://github.com/AlexandreRouma/SDRPlusPlus/issues/1340)"
" [Add support for RTL-SDR Blog V4 #1170](https://github.com/AlexandreRouma/SDRPlusPlus/issues/1170)"

I followed the instruction at:
"[RTL-SDR Blog V4 Users Guide ](https://www.rtl-sdr.com/V4/)" but RTLsdr v4 still doesn't work.  All I get is white noise.
The instructions are the same as in the solution at https://osmocom.org/projects/rtl-sdr/wiki.

Here are the instruction at RTL-SDR Blog V4 Users Guide 
Linux (Debian)

On Linux the procedure is also simple quite simple and just involves removing any existing RTL-SDR drivers, and installing our version.

    Purge the previous driver:

    sudo apt purge ^librtlsdr
    sudo rm -rvf /usr/lib/librtlsdr* /usr/include/rtl-sdr* /usr/local/lib/librtlsdr* /usr/local/include/rtl-sdr* /usr/local/include/rtl_* /usr/local/bin/rtl_* 

    Install the RTL-SDR Blog drivers:

    sudo apt-get install libusb-1.0-0-dev git cmake pkg-config
    git clone https://github.com/rtlsdrblog/rtl-sdr-blog
    cd rtl-sdr-blog
    mkdir build
    cd build
    cmake ../ -DINSTALL_UDEV_RULES=ON
    make
    sudo make install
    sudo cp ../rtl-sdr.rules /etc/udev/rules.d/
    sudo ldconfig

    Blacklist the DVB-T TV drivers.

    echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf

    Reboot

wptracy, 2024-03-27:
Never mind.  It must have just been a bad antenna connection.

I swapped the v3 and v4 between the two raspberry pi's and they both worked.

I swapped them back and they both worked.

closed 2024-03-27 by wptracy

piadam, 2024-05-07:
I am having the same problem. Too bad I don't have a V3 to swap the V4 with. The V4 works great on my MacBook Air but just produces white noise on my bookworm Pi's 3B and 5.

vali20252025, 2025-03-11:
Good evening, I've been trying different compilation options for sdr v4 and Pi5 but I can't find anything concrete, repeatable to be able to use the Pi board and rtl v4 for remote connection via sdrpp. Any help from you would be greatly appreciated.

Brabudu, 2025-05-07:
Same problem: my RTL-SDR V4 works perfectly in windows but does not work on raspberry pi400, it's only white noise. RTL-SDR V3 instead work on both Raspbian and Windows.