AlexandreRouma/SDRPlusPlus #1368: FM MPX Audio Output

KubaPro010, 2024-03-30:
Would it would be ever possible to add a audio fm multiplex output? to for example analyze the spectrum or feed it into RDS Spy? basically something like SDRConsole has

AlexandreRouma, 2024-03-30:
If you disable low-pass and deemphasis the demodulator outputs the raw MPX.

KubaPro010, 2024-03-30:
Doesn't seem to work correctly, here are my settings:
[image]
why do i think it doesn't work correctly? becuase of this, captured with a another instance of sdr++
[image]

KubaPro010, 2024-03-30:
Now i saw that i had a loopback, so sdr++ demod to sdr++ input, but without that still not good mpx output
[image]

AlexandreRouma, 2024-03-30:
stereo needs to be off also forgot about it

KubaPro010, 2024-03-30:
yeah, okay, seems to work now, i guess thanks for the tips