AlexandreRouma/SDRPlusPlus #1371: Start fail on uConsole-a06

Jamel-Tayeb, 2024-04-02:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: ARM Cortex-A53
- RAM: 4 GB
- GPU: NA
- SDR: (Remote or local? If remote, what protocol?)
local 

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)

Linux clockworkpi-a06 5.15.119-rockchip64 #trunk SMP PREEMPT Sat Jul 1 19:53:59 CST 2023 aarch64 aarch64 aarch64 GNU/Linux

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04 LTS
Release:	22.04
Codename:	jammy

- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

2266798 Apr  1 14:34 sdrpp_ubuntu_jammy_amd64.deb

**Bug Description**
A clear description of the bug.

start fails

sdrpp
Dynarec for ARM64, with extension: ASIMD AES CRC32 PMULL PageSize:4096
Box64 with Dynarec v0.1.9 6392550 built on Jun 11 2022 19:30:12
BOX64_LD_LIBRARY_PATH: ./:lib/:lib64/:x86_64/:bin64/:libs64/:/usr/lib/sdrpp/:/usr/lib/aarch64-linux-gnu/
BOX64_PATH: ./:bin/
Counted 50 Env var
Looking for /usr/bin/sdrpp
Using emulated /usr/lib/libsdrpp_core.so
Using native(wrapped) libc.so.6
Using native(wrapped) ld-linux-x86-64.so.2
Using native(wrapped) libpthread.so.0
Using native(wrapped) librt.so.1
Error loading needed lib libglfw.so.3
Error loading one of needed lib
Error initializing needed lib libsdrpp_core.so
Error loading one of needed lib
Error: loading needed libs in elf /usr/bin/sdrpp

**Steps To Reproduce**
1. follow documentation install steps, all steps are ok
2. run sdrpp - fails with missing libglfw (see above)
the lib is present:
sudo find / -name "*glfw*"
/var/lib/dpkg/info/libglfw3:arm64.shlibs
/var/lib/dpkg/info/libglfw3:arm64.triggers
/var/lib/dpkg/info/libglfw3:arm64.symbols
/var/lib/dpkg/info/libglfw3:arm64.md5sums
/var/lib/dpkg/info/libglfw3:arm64.list
/var/lib/dpkg/info/libglfw3-dev:arm64.list
/var/lib/dpkg/info/libglfw3-dev:arm64.md5sums
/usr/lib/aarch64-linux-gnu/pkgconfig/glfw3.pc
/usr/lib/aarch64-linux-gnu/libglfw.so.3.3
/usr/lib/aarch64-linux-gnu/cmake/glfw3
/usr/lib/aarch64-linux-gnu/cmake/glfw3/glfw3Config.cmake
/usr/lib/aarch64-linux-gnu/cmake/glfw3/glfw3Targets-none.cmake
/usr/lib/aarch64-linux-gnu/cmake/glfw3/glfw3Targets.cmake
/usr/lib/aarch64-linux-gnu/cmake/glfw3/glfw3ConfigVersion.cmake
/usr/lib/aarch64-linux-gnu/libglfw.so
/usr/lib/aarch64-linux-gnu/libglfw.so.3
/usr/include/GLFW/glfw3.h
/usr/include/GLFW/glfw3native.h
/usr/share/doc/libglfw3
/usr/share/doc/libglfw3-dev

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`

sdrpp -c
Dynarec for ARM64, with extension: ASIMD AES CRC32 PMULL PageSize:4096
Box64 with Dynarec v0.1.9 6392550 built on Jun 11 2022 19:30:12
BOX64_LD_LIBRARY_PATH: ./:lib/:lib64/:x86_64/:bin64/:libs64/:/usr/lib/sdrpp/:/usr/lib/aarch64-linux-gnu/
BOX64_PATH: ./:bin/
Counted 50 Env var
Looking for /usr/bin/sdrpp
argv[1]="-c"
Using emulated /usr/lib/libsdrpp_core.so
Using native(wrapped) libc.so.6
Using native(wrapped) ld-linux-x86-64.so.2
Using native(wrapped) libpthread.so.0
Using native(wrapped) librt.so.1
Error loading needed lib libglfw.so.3
Error loading one of needed lib
Error initializing needed lib libsdrpp_core.so
Error loading one of needed lib
Error: loading needed libs in elf /usr/bin/sdrpp

* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

I tried adding the lib to LD_LIBRARY_PATF and BOX64_LD_LIBRARY_PATH with no success

AlexandreRouma, 2024-04-02:
Why are you expecting an AMD64 executable to run on Arm? It's right in the name of the file you downloaded:
"sdrpp_ubuntu_jammy_***amd64***.deb"

Build it from source.

closed 2024-04-02 by AlexandreRouma

Jamel-Tayeb, 2024-04-02:
good point :)

jtayeb, 2024-04-02:
Done, works like a charm - thanks for pointing out my blindness :)