AlexandreRouma/SDRPlusPlus #1372: Contents of frequency_manager_config.json are deleted

rogerxxxx, 2024-04-04:
After editing the frequency_manager_config.json (eg. $HOME/.config/sdrpp) with a text editor and creating syntax errors within the file, upon subsequent start of SDRPlusPlus, SDRPlusPlus likely detects the anomalies and simply wipes/deletes the contents of the frequency_manager_config.json file.

SDRPlusPlus should instead make a dated/timed back-up of the frequency_manager_config.json file.

Hams, Amateur Radio enthusiasts, and shortwave listeners will likely have many essential frequencies saved within the file.  Editing the file using a text editor is always far easier than using a graphical frontend, at the risk of mistakes.

WORKAROUND: For the time being, just redundantly making text file backups, treating all text files as volatile, as we did during the 1990s and prior.

AlexandreRouma, 2024-04-04:
You're not supposed to edit the config files by hand and it will not be supported.
If you decide to do it anyway, you are responsible for backing them up.

Also I'm not sure exactly how editing JSON by hand can be faster than entering the text fields only and pressing one button.
If you need to add a large number of entries at once, that's what the import/export feature is for.

closed 2024-04-04 by AlexandreRouma

rogerxxxx, 2024-04-05:
Using vim, can easily copy/paste sections.  (eg. html, xml, json, ...)  Far easier than remove hand for keyboard for mouse click, type, remove hands for mouse click, type, ...  Using a grpahical text editor only using a mouse would likely be far more difficult.

The likely reason for wiping/deleting the file, there's likely no fallback in the source code when detecting an error condition within the frequency manager file.  Probably should catch it, and at least copy the existing poorly formatted file to a backup file.  (eg. frequency_manager.json.20240305-033830)

Another method is to send an error to console or GUI, requesting the user to backup the file before manually removing the file.

Think most Linux/Unix based programs usually perform one or the other.