AlexandreRouma/SDRPlusPlus #1374: error: 'class POCSAGDSP' has no member named 'detune'

rogerxxxx, 2024-04-05:
```
[ 84%] Building CXX object decoder_modules/pager_decoder/CMakeFiles/pager_decoder.dir/src/main.cpp.o
In file included from /home/roger/src/SDRPlusPlus.mine/core/src/signal_path/../dsp/channel/rx_vfo.h:2,
                 from /home/roger/src/SDRPlusPlus.mine/core/src/signal_path/vfo_manager.h:2,
                 from /home/roger/src/SDRPlusPlus.mine/core/src/gui/main_window.h:6,
                 from /home/roger/src/SDRPlusPlus.mine/core/src/gui/gui.h:7,
                 from /home/roger/src/SDRPlusPlus.mine/decoder_modules/pager_decoder/src/main.cpp:5:
/home/roger/src/SDRPlusPlus.mine/core/src/signal_path/../dsp/channel/frequency_xlator.h: In member function 'int dsp::channel::FrequencyXlator::process(int, const dsp::complex_t*, dsp::complex_t*)':
/home/roger/src/SDRPlusPlus.mine/core/src/signal_path/../dsp/channel/frequency_xlator.h:44:13: warning: 'volk_32fc_s32fc_x2_rotator_32fc' is deprecated [-Wdeprecated-declarations]
   44 |             volk_32fc_s32fc_x2_rotator_32fc((lv_32fc_t*)out, (lv_32fc_t*)in, phaseDelta, &phase, count);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/roger/src/SDRPlusPlus.mine/core/src/dsp/stream.h:5,
                 from /home/roger/src/SDRPlusPlus.mine/core/src/gui/main_window.h:5:
/usr/include/volk/volk.h:1751:46: note: declared here
 1751 | extern VOLK_API p_32fc_s32fc_x2_rotator_32fc volk_32fc_s32fc_x2_rotator_32fc __attribute__((deprecated));
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/roger/src/SDRPlusPlus.mine/decoder_modules/pager_decoder/src/main.cpp:11:
/home/roger/src/SDRPlusPlus.mine/decoder_modules/pager_decoder/src/pocsag/decoder.h: In member function 'virtual void POCSAGDecoder::showMenu()':

/home/roger/src/SDRPlusPlus.mine/decoder_modules/pager_decoder/src/pocsag/decoder.h:49:17: error: 'class POCSAGDSP' has no member named 'detune'
   49 |             dsp.detune();
      |                 ^~~~~~
make[2]: *** [decoder_modules/pager_decoder/CMakeFiles/pager_decoder.dir/build.make:90: decoder_modules/pager_decoder/CMakeFiles/pager_decoder.dir/src/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1913: decoder_modules/pager_decoder/CMakeFiles/pager_decoder.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

Modified  CMakeLists.txt, I do not have the following devices, disabled them.  The libad9361 looks like a 2G/3G transceiver base station, nor was the libad9361-dev package available on Void Linux.  Doesn't look like the above build error was the source of the apparent dependency error, regardless, including changes to CMakeLists.txt

$ diff -urN ../CMakeLists.txt.orig ../CMakeLists.txt
-option(OPT_BUILD_AIRSPY_SOURCE "Build Airspy Source Module (Dependencies: libairspy)" ON)                                                                                                         
+option(OPT_BUILD_AIRSPY_SOURCE "Build Airspy Source Module (Dependencies: libairspy)" OFF)

-option(OPT_BUILD_HACKRF_SOURCE "Build HackRF Source Module (Dependencies: libhackrf)" ON)                                                                                                         
+option(OPT_BUILD_HACKRF_SOURCE "Build HackRF Source Module (Dependencies: libhackrf)" OFF) 

-option(OPT_BUILD_PLUTOSDR_SOURCE "Build PlutoSDR Source Module (Dependencies: libiio, libad9361)" ON)                                                                                             
+option(OPT_BUILD_PLUTOSDR_SOURCE "Build PlutoSDR Source Module (Dependencies: libiio, libad9361)" OFF)

Error is within the internal pocsag pager decoder sources, likely not finding a header file.

WORKAROUND:
-option(OPT_BUILD_PAGER_DECODER "Build the pager decoder module (no dependencies required)" ON)
+option(OPT_BUILD_PAGER_DECODER "Build the pager decoder module (no dependencies required)" OFF)

... building/compiling is continuing past the compiling previous error. 

```

AlexandreRouma, 2024-04-05:
already been fixed an hour ago, pull again.

closed 2024-04-05 by AlexandreRouma

rogerxxxx, 2024-04-05:
wow. that was quick.

git pull, confirmed some deletions to the specific pager .h file... removed my modification to the cmakelists.txt omitting the pager module build, make clean && make shows everything builds as expected, without the pager module compile error.

fixed.