AlexandreRouma/SDRPlusPlus #1377: Raspberry Pi Install Guide

TeddJohnson, 2024-04-07:
Hoping this will help anyone else in the future trying this out.

If you're installing on Raspberry Pi, you likely don't want to use the nightly images. 

`aarch64` is NOT the same as `amd64` and the ` sdrpp_raspios_bullseye_armhf.deb ` package does not work on 64-bit OSes (which is likely what you have if you have a more recent raspberry pi).

Instead, follow the `Building on Linux / BSD` Section of the ReadMe. I had already installed all the linux dependancies from the previous attempt, but this may still help you:
```sudo apt install libfftw3-dev libglfw3-dev libvolk2-dev libzstd-dev libsoapysdr-dev libairspyhf-dev libiio-dev libad9361-dev librtaudio-dev libhackrf-dev``` I also had to install `sudo apt install librtlsdr-dev cmake make build-essential`

I had it not install the airspy dependency (Because I don't need it for my RTL-SDR V4), but if you need it you'll have to figure that one out (That's the cname -D switch below)

Then:

```mkdir sdrpp-build
git clone https://github.com/AlexandreRouma/SDRPlusPlus.git
cd sdrpp-build
cmake ../SDRPlusPlus -DOPT_BUILD_AIRSPY_SOURCE=False
make -j `nproc`
cd ../SDRPlusPlus
sh ./create_root.sh
cd ../sdrpp-build
sudo make install```

you may not need this bit:
```cd ../SDRPlusPlus
sh ./create_root.sh
cd ../sdrpp-build```

but I already ran it so I can't confirm if it's not necessary.

now sdrpp runs great (currently I'm running it in VNC, but I've seen someone say they had issues with that. I haven't gotten far enough along to know)

tim-rtd, 2024-04-13:
Perfect copy and paste procedure.

AlexandreRouma, 2024-04-16:
Firstly, this place is for reporting bugs and requesting feature, not really to write posts like that, so i'll close this issue.

Secondly, beware that using those commands will give you an incomplete build compared to nightly build. To get the same modules, you must enable them following the "built in releases" column of the module list visible in the readme.

closed 2024-04-16 by AlexandreRouma

TeddJohnson, 2024-05-01:
Thank you for the reply and frankly for building the software and sharing it at all. I really do appreciate the momentous effort it takes

1. I've seen numerous people struggle with getting it to run on their raspberry pi, and just wanted to help. I think it's a pretty basic mistake (thinking AMD64 works for ARM64 and not being able to install the ARMHF.deb package on a 64-bit Rpi OS) so I thought I'd post this to help anyone else searching. If there's somewhere better to post it, please let me know.

2. I changed the instructions in the original post to address using the nightly build. in the ~3 weeks since I posted this I've learned a bit more, so hopefully these instructions provide users with the more up-to-date nightly build.