AlexandreRouma/SDRPlusPlus #1379: Crash on Alpine Linux in Radio module

eddsalkield, 2024-04-11:
**Hardware**
- CPU: Intel(R) Core(TM) i7-5600U CPU @ 2.60GHz
- RAM: MemTotal: 16271040 kB
- GPU: Intel HD Graphics 5500
- SDR: N/A

**Software**
- Operating System: Alpine edge
- SDR++: 1.1.0 + commit #e118598f5763d29862cc3ff143e23a36868bffe4

**Bug Description**
`sdrpp` crashes on startup when loading the Radio module with exception `Illegal instruction`.

**Steps To Reproduce**
`sdrpp`

**Additional info**
Backtrace:
```
Starting program: /usr/bin/sdrpp 
[11/04/2024 16:23:59.000] [INFO] SDR++ v1.1.0
[11/04/2024 16:23:59.000] [INFO] Loading config
[New LWP 13502]
[New LWP 13513]
[New LWP 13514]
[New LWP 13515]
[New LWP 13516]
[11/04/2024 16:23:59.000] [INFO] Using OpenGL 3.0
[11/04/2024 16:23:59.000] [INFO] Loading icons
[11/04/2024 16:23:59.000] [INFO] Loading band plans
[11/04/2024 16:23:59.000] [INFO] Loading band plans color table
[11/04/2024 16:23:59.000] [ERROR] Menu element is missing name key
[New LWP 13517]
[New LWP 13518]
[New LWP 13519]
[New LWP 13520]
[New LWP 13521]
[New LWP 13522]
[11/04/2024 16:23:59.000] [INFO] Loading modules
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/scanner.so
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_sink.so
[New LWP 13523]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/pager_decoder.so
[New LWP 13524]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[New LWP 13525]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[New LWP 13526]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[New LWP 13527]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[New LWP 13528]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/recorder.so
[New LWP 13529]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[New LWP 13530]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[New LWP 13531]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/radio.so
[New LWP 13532]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[New LWP 13533]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/iq_exporter.so
[New LWP 13534]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[New LWP 13535]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_source.so
[New LWP 13536]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[New LWP 13537]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_client.so
[New LWP 13538]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[New LWP 13539]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/usrp_source.so
[New LWP 13540]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[New LWP 13541]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[New LWP 13542]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[New LWP 13543]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[New LWP 13544]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/file_source.so
[New LWP 13545]
[11/04/2024 16:23:59.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_source.so
[New LWP 13546]
[11/04/2024 16:23:59.000] [INFO] Initializing Airspy Source (airspy_source)
[New LWP 13547]
[11/04/2024 16:23:59.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[LWP 13547 exited]
[New LWP 13548]
[11/04/2024 16:23:59.000] [INFO] Initializing Audio Sink (audio_sink)
[LWP 13548 exited]
[11/04/2024 16:23:59.000] [INFO] Initializing Audio Source (audio_source)
[New LWP 13549]
[LWP 13549 exited]
[New LWP 13550]
[LWP 13550 exited]
[New LWP 13551]
[LWP 13551 exited]
[New LWP 13552]
[LWP 13552 exited]
[New LWP 13553]
[LWP 13553 exited]
[LWP 13521 exited]
[LWP 13522 exited]
[LWP 13520 exited]
[New LWP 13554]
[New LWP 13555]
[New LWP 13556]
[11/04/2024 16:24:00.000] [INFO] New DSP samplerate: 48000.000000 (source samplerate is 48000.000000)
[11/04/2024 16:24:00.000] [INFO] Initializing BladeRF Source (bladerf_source)
[11/04/2024 16:24:00.000] [ERROR] Module 'bladerf_source' doesn't exist
[11/04/2024 16:24:00.000] [INFO] Initializing File Source (file_source)
[11/04/2024 16:24:00.000] [INFO] Initializing Frequency Manager (frequency_manager)
[11/04/2024 16:24:00.000] [INFO] Initializing HackRF Source (hackrf_source)
[New LWP 13557]
[11/04/2024 16:24:00.000] [INFO] Initializing Hermes Source (hermes_source)
[11/04/2024 16:24:00.000] [INFO] Initializing LimeSDR Source (limesdr_source)
[11/04/2024 16:24:00.000] [ERROR] Module 'limesdr_source' doesn't exist
[11/04/2024 16:24:00.000] [INFO] Initializing Network Sink (network_sink)
[11/04/2024 16:24:00.000] [INFO] Initializing PerseusSDR Source (perseus_source)
[11/04/2024 16:24:00.000] [ERROR] Module 'perseus_source' doesn't exist
[11/04/2024 16:24:00.000] [INFO] Initializing PlutoSDR Source (plutosdr_source)
[11/04/2024 16:24:00.000] [ERROR] Module 'plutosdr_source' doesn't exist
[11/04/2024 16:24:00.000] [INFO] Initializing RFspace Source (rfspace_source)
[11/04/2024 16:24:00.000] [INFO] Initializing RTL-SDR Source (rtl_sdr_source)
[11/04/2024 16:24:00.000] [INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[11/04/2024 16:24:00.000] [INFO] Initializing Radio (radio)
[Resamp] predec: 8192, interp: 25, decim: 6, inacc: 0.000000%, taps: 1900
[New LWP 13558]
[LWP 13519 exited]
[New LWP 13559]
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[New LWP 13560]
[LWP 13560 exited]
[New LWP 13561]
[LWP 13561 exited]
[New LWP 13562]
[LWP 13562 exited]
[New LWP 13563]
[LWP 13563 exited]
[New LWP 13564]
[LWP 13564 exited]
[Resamp] predec: 32, interp: 5000, decim: 7813, inacc: 0.000000%, taps: 593750
[LWP 13559 exited]
[Resamp] predec: 8192, interp: 125, decim: 24, inacc: 0.000000%, taps: 9500

Thread 1 "sdrpp" received signal SIGILL, Illegal instruction.
0x00007ffff785ab7b in ?? () from /usr/lib/libsdrpp_core.so
(gdb) bt
#0  0x00007ffff785ab7b in ?? () from /usr/lib/libsdrpp_core.so
#1  0x00007fffd0b53e51 in RadioModule::selectDemod(demod::Demodulator*) ()
   from /usr/lib/sdrpp/plugins/radio.so
#2  0x00007fffd0b56df1 in RadioModule::selectDemodByID(RadioModule::DemodID) ()
   from /usr/lib/sdrpp/plugins/radio.so
#3  0x00007fffd0b5916f in RadioModule::RadioModule(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) () from /usr/lib/sdrpp/plugins/radio.so
#4  0x00007fffd0b2a6e6 in _CREATE_INSTANCE_ () from /usr/lib/sdrpp/plugins/radio.so
#5  0x00007ffff79ac251 in ModuleManager::createInstance(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) () from /usr/lib/libsdrpp_core.so
#6  0x00007ffff78b9ee0 in MainWindow::init() () from /usr/lib/libsdrpp_core.so
#7  0x00007ffff788965f in sdrpp_main(int, char**) () from /usr/lib/libsdrpp_core.so
#8  0x00007ffff7f786d1 in libc_start_main_stage2 (main=0x555555555030, argc=1, 
    argv=0x7fffffffe8c8) at src/env/__libc_start_main.c:95
#9  0x000055555555504c in ?? ()
#10 0x0000000000000001 in ?? ()
#11 0x00007fffffffeb98 in ?? ()
#12 0x0000000000000000 in ?? ()
```

AlexandreRouma, 2024-04-12:
Build and install a debug build and get a more precise backtrace.

eddsalkield, 2024-04-16:
I've installed a debug build, and got the following trace:
```
Thread 1 "sdrpp" received signal SIGILL, Illegal instruction.
0x00007ffff79c806a in dsp::filter::FIR<dsp::complex_t, float>::setTaps(dsp::tap<float>&) () from /usr/lib/libsdrpp_core.so
(gdb) bt
#0  0x00007ffff79c806a in dsp::filter::FIR<dsp::complex_t, float>::setTaps(dsp::tap<float>&) () from /usr/lib/libsdrpp_core.so
#1  0x00007ffff79d64ef in dsp::channel::RxVFO::setOutSamplerate(double, double) () from /usr/lib/libsdrpp_core.so
#2  0x00007ffff79d52c0 in VFOManager::VFO::setSampleRate(double, double) () from /usr/lib/libsdrpp_core.so
#3  0x00007fffd0a83766 in RadioModule::selectDemod(demod::Demodulator*) () from /usr/lib/sdrpp/plugins/radio.so
#4  0x00007fffd0a83a98 in RadioModule::selectDemodByID(RadioModule::DemodID) () from /usr/lib/sdrpp/plugins/radio.so
#5  0x00007fffd0a85719 in RadioModule::RadioModule(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) () from /usr/lib/sdrpp/plugins/radio.so
#6  0x00007fffd0a65316 in _CREATE_INSTANCE_ () from /usr/lib/sdrpp/plugins/radio.so
#7  0x00007ffff79c016f in ModuleManager::createInstance(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) () from /usr/lib/libsdrpp_core.so
#8  0x00007ffff790221c in MainWindow::init() () from /usr/lib/libsdrpp_core.so
#9  0x00007ffff78e92b2 in sdrpp_main(int, char**) () from /usr/lib/libsdrpp_core.so
#10 0x000055555555517f in ?? ()
#11 0x00007ffff7f786d1 in libc_start_main_stage2 (main=0x555555555175, argc=1, argv=0x7fffffffe8c8) at src/env/__libc_start_main.c:95
#12 0x0000555555555046 in ?? ()
#13 0x0000000000000001 in ?? ()
#14 0x00007fffffffeb97 in ?? ()
#15 0x0000000000000000 in ?? ()
```

eddsalkield, 2024-04-16:
For some reason the symbols got stripped in the previous reply, here's the complete backtrace:
```
Thread 1 "sdrpp" received signal SIGILL, Illegal instruction.
dsp::filter::FIR<dsp::complex_t, float>::setTaps (this=this@entry=0x7fffd1080818, taps=...)
    at /home/edd/dev/aports/testing/sdr++/src/SDRPlusPlus-e118598f5763d29862cc3ff143e23a36868bffe4/core/src/signal_path/../dsp/multirate/../filter/fir.h:47
47	               memcpy(&buffer[_taps.size - oldTC], buffer, (oldTC - 1) * sizeof(D));
(gdb) bt full
#0  dsp::filter::FIR<dsp::complex_t, float>::setTaps (this=this@entry=0x7fffd1080818, 
    taps=...)
    at /home/edd/dev/aports/testing/sdr++/src/SDRPlusPlus-e118598f5763d29862cc3ff143e23a36868bffe4/core/src/signal_path/../dsp/multirate/../filter/fir.h:47
        __func__ = "setTaps"
        lck = {_M_device = @0x7fffd1080828}
        oldTC = 76
#1  0x00007ffff79d64ef in dsp::channel::RxVFO::setOutSamplerate (this=0x7fffd10800b0, 
    outSamplerate=outSamplerate@entry=250000, bandwidth=bandwidth@entry=150000)
    at /home/edd/dev/aports/testing/sdr++/src/SDRPlusPlus-e118598f5763d29862cc3ff143e23a36868bffe4/core/src/signal_path/../dsp/channel/rx_vfo.h:55
        __func__ = "setOutSamplerate"
        lck = {_M_device = @0x7fffd10800c0}
#2  0x00007ffff79d52c0 in VFOManager::VFO::setSampleRate (this=0x7fffcf9e2d70, 
    sampleRate=sampleRate@entry=250000, bandwidth=150000)
    at /home/edd/dev/aports/testing/sdr++/src/SDRPlusPlus-e118598f5763d29862cc3ff143e23a36868bffe4/core/src/signal_path/vfo_manager.cpp:52
No locals.
#3  0x00007fffd0c77766 in RadioModule::selectDemod (this=this@entry=0x7fffc898c4c0, 
    demod=<optimized out>)
    at /home/edd/dev/aports/testing/sdr++/src/SDRPlusPlus-e118598f5763d29862cc3ff143e23a36868bffe4/decoder_modules/radio/src/radio_module.h:421
        ifSamplerate = 250000
#4  0x00007fffd0c77a98 in RadioModule::selectDemodByID (this=this@entry=0x7fffc898c4c0, 
    id=<optimized out>)
    at /home/edd/dev/aports/testing/sdr++/src/SDRPlusPlus-e118598f5763d29862cc3ff143e23a36868bffe4/decoder_modules/radio/src/radio_module.h:329
        startTime = {__d = {__r = 1713297636490676523}}
        demod = <optimized out>
        endTime = {__d = {__r = 0}}
#5  0x00007fffd0c79719 in RadioModule::RadioModule (this=0x7fffc898c4c0, name=...)
    at /home/edd/dev/aports/testing/sdr++/src/SDRPlusPlus-e118598f5763d29862cc3ff143e23a36868bffe4/decoder_modules/radio/src/radio_module.h:97
        created = <optimized out>
        created = <optimized out>
#6  0x00007fffd0c59316 in _CREATE_INSTANCE_ (name=...)
    at /home/edd/dev/aports/testing/sdr++/src/SDRPlusPlus-e118598f5763d29862cc3ff143e23a36868bffe4/decoder_modules/radio/src/main.cpp:19
No locals.
#7  0x00007ffff79c016f in ModuleManager::createInstance (
    this=0x7ffff7a74280 <core::moduleManager>, name=..., module=...)
    at /home/edd/dev/aports/testing/sdr++/src/SDRPlusPlus-e118598f5763d29862cc3ff143e23a36868bffe4/core/src/module.cpp:102
        maxCount = -1
        inst = {module = {handle = 0x7fffe1716d40, info = 0x7fffd0c98d20 <_INFO_>, 
            init = 0x7fffd0c5937d <_INIT_()>, 
--Type <RET> for more, q to quit, c to continue without paging--c
            createInstance = 0x7fffd0c592d7 <_CREATE_INSTANCE_(std::string)>, 
            deleteInstance = 0x7fffd0c592b7 <_DELETE_INSTANCE_(void*)>, 
            end = 0x7fffd0c59115 <_END_()>}, instance = 0x7fffffffe340}
#8  0x00007ffff790221c in MainWindow::init (this=0x7ffff7e47400 <gui::mainWindow>)
    at /home/edd/dev/aports/testing/sdr++/src/SDRPlusPlus-e118598f5763d29862cc3ff143e23a36868bffe4/core/src/gui/main_window.cpp:141
        mod = {
          _M_dataplus = {<std::allocator<char>> = {<std::__new_allocator<char>> = {<No data fields>}, <No data fields>}, _M_p = 0x7fffffffe490 "radio"}, _M_string_length = 5, {
            _M_local_buf = "radio\000p_source\000", 
            _M_allocated_capacity = 6876997109504303474}}
        enabled = true
        _module = <optimized out>
        __for_range = @0x7fffffffe3a8: {container = @0x7ffff6ba94f0}
        __for_begin = {anchor = {m_object = 0x7ffff6ba94f0, m_it = {object_iterator = {
                _M_node = 0x7ffff6b83510}, array_iterator = {_M_current = 0x0}, 
              primitive_iterator = {static begin_value = 0, static end_value = 1, 
                m_it = -9223372036854775808}}}, array_index = 16, array_index_last = 0, 
          array_index_str = {
            _M_dataplus = {<std::allocator<char>> = {<std::__new_allocator<char>> = {<No data fields>}, <No data fields>}, _M_p = 0x7fffffffe560 "0"}, _M_string_length = 1, {
              _M_local_buf = "0\000\316\341\377\177\000\000\202\r\370\367\377\177\000", 
              _M_allocated_capacity = 140736981762096}}, empty_str = {
            _M_dataplus = {<std::allocator<char>> = {<std::__new_allocator<char>> = {<No data fields>}, <No data fields>}, _M_p = 0x7fffffffe580 ""}, _M_string_length = 0, {
              _M_local_buf = "\000\000\000\000\000\000\000\000\210\346\377\377\377\177\000", 
              _M_allocated_capacity = 0}}}
        __for_end = {anchor = {m_object = 0x7ffff6ba94f0, m_it = {object_iterator = {
[... 35 lines cut ...]
        serverMode = <optimized out>
        root = {
          _M_dataplus = {<std::allocator<char>> = {<std::__new_allocator<char>> = {<No data fields>}, <No data fields>}, _M_p = 0x7ffff6ee3990 "/home/edd/.config/sdrpp"}, 
          _M_string_length = 23, {
[... cut ...]
        bandColors = {m_type = nlohmann::detail::value_t::object, m_value = {
            object = 0x7ffff6b57ae0, array = 0x7ffff6b57ae0, string = 0x7ffff6b57ae0, 
            binary = 0x7ffff6b57ae0, boolean = 224, number_integer = 140737332476640, 
            number_unsigned = 140737332476640, number_float = 6.9533481064045384e-310}}
        biRes = <optimized out>
#10 0x000055555555517f in main (argc=<optimized out>, argv=<optimized out>)
    at /home/edd/dev/aports/testing/sdr++/src/SDRPlusPlus-e118598f5763d29862cc3ff143e23a36868bffe4/src/main.cpp:5
No locals.
```

AlexandreRouma, 2024-04-16:
How did you compile it? The fact it crashes on a memcpy with an illegal instruction exception sounds a lot like it was compiled for vector extensions that your CPU does not support.

eddsalkield, 2024-04-16:
I compiled it using the following:

```
    cmake -B build -DCMAKE_BUILD_TYPE=Debug -DOPT_BUILD_USRP_SOURCE=ON \
    	-DOPT_BUILD_PLUTOSDR_SOURCE=0
```

I slightly changed a CMakeList to disable asan, which is not supported under musl libc:

```
diff old/core/libcorrect/CMakeLists.txt new/core/libcorrect/CMakeLists.txt
--- old/core/libcorrect/CMakeLists.txt
+++ new/core/libcorrect/CMakeLists.txt
@@ -17,14 +17,8 @@
 set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wpedantic")
 endif()
 if(CMAKE_BUILD_TYPE STREQUAL "Debug")
-    # On android, keep optimisations and don't use asan
-    if (ANDROID)
-      set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -g3 -O3")
-      set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -Wl,")
-    else()
-      set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -g3 -O0 -fsanitize=address")
-      set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -Wl,-no_pie,")
-    endif()
+    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -g3 -O0")
+    set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -Wl,-no_pie,")
 else()
   if("${CMAKE_C_COMPILER_ID}" STREQUAL "Clang")
       set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -O2")
```

Further poking in the assembly layout of GDB shows that the precise CPU instruction that causes the failure happens on instruction `ud2`.

This seems to be a [specifically undefined](https://web.archive.org/web/20120425092211/siyobik.info.gf/main/reference/instruction/UD2) instruction, which is confusing.

```
|  0x7ffff79c805d <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+263>           mov    $0x0,%edx                                                                                      │
│   0x7ffff79c8062 <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+268>           call   *0x9ec28(%rip)        # 0x7ffff7a66c90                                                         │
│   0x7ffff79c8068 <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+274>           jmp    0x7ffff79c8005 <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+175>                  │
│  >0x7ffff79c806a <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+276>           ud2                                                                                                   │
│   0x7ffff79c806c <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+278>           mov    %r12,%rdi
```

AlexandreRouma, 2024-04-16:
It's preceded by a non conditional jump instructions so the undefined instruction after it is not surprising.

You should look at what the code its jumping to looks like.

eddsalkield, 2024-04-16:
afaict, we're jumping here:

```
│   0x7ffff79c7ffd <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+167>   jb     0x7ffff79c8010 <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+186>                          │
│   0x7ffff79c7fff <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+169>   call   *0x9ed9b(%rip)        # 0x7ffff7a66da0                                                                 │
│   0x7ffff79c8005 <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+175>   mov    %rbx,%rdi                                                                                              │
│   0x7ffff79c8008 <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+178>   call   *0xa0b02(%rip)        # 0x7ffff7a68b10                                                                 │
│   0x7ffff79c800e <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+184>   jmp    0x7ffff79c806c <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+278>                          │
│   0x7ffff79c8010 <_ZN3dsp6filter3FIRINS_9complex_tEfE7setTapsERNS_3tapIfEE+186>   ud2                                                                                                           │
```

Looks like it should be jumping back to `0x7ffff79c806c` after the call, but somehow we end up at `0x7ffff79c806a` instead, which is our `ud2`. I can't think of why - do you have any ideas?

Thanks so much for your help, btw

AlexandreRouma, 2024-04-16:
I'm not sure. I think the only way to figure this out is to put a breakpoint a bit before when it happens and step instruction by instruction nothing the exact program flow along the way.

eddsalkield, 2024-04-16:
More digging with gdb revealed that the following section of `/usr/include/fortify/string.h` from the [fortify-headers](http://git.2f30.org/fortify-headers/) package was detecting overlapping pointers for the memcpy, which is UB.

The fix would be to use std::copy, which is defined for when the destination iterator is outside the range to be copied, even if the ranges overlap.

I've fixed it in this patch https://github.com/eddsalkield/SDRPlusPlus/commit/4ba2ad3f4bb1c0a9da5abc89109848851eb7e861, which I was going to submit as a PR, but apparently that's unwelcome. What do you think of the patch?

AlexandreRouma, 2024-04-16:
Ngl, that package should probably not just outright crash software with an illegal instruction... This is probably the least debuggable way to report such a fault...

`std::copy` is no good due to C++ bloat. The correct solution is to use `memmove` instead. I'll push a fix.

AlexandreRouma, 2024-04-16:
Should be fixed, let me know if not.

closed 2024-04-16 by AlexandreRouma

eddsalkield, 2024-04-17:
We're now starting correctly, thanks!