AlexandreRouma/SDRPlusPlus #1380: Cannot access the High DPI menu

WRJK988, 2024-04-12:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Onn Tablet
- RAM: not sure
- GPU: 
- SDR: RTL-SDR.COM V3

**Software**
- Operating System: Android version 9
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).
- I cant see the SDR++ version or build date due to the same reason.
[image]

**Bug Description**
Cannot access the High DPI menu due to the screen sizing

**Steps To Reproduce**
1. ...
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
[image]

**Additional info**
Add any other relevant information.

AlexandreRouma, 2024-04-12:
Expand the menu by dragging its side (between menu and waterfall). Works the exact same way as it does on desktop.

closed 2024-04-12 by AlexandreRouma

WRJK988, 2024-04-12:
Thank you! I tried that a few times, but that drag line is picky. That
fixed it!