AlexandreRouma/SDRPlusPlus #1381: Spectrum/waterfall disappears after computer nap

JohBod, 2024-04-14:
**Hardware**
- CPU: 
[image]

- RAM: 16GB
- GPU: 
[image]

- SDR: (Remote or local? If remote, what protocol?)
Local. Airspy HF+ (old, "dual port" version)

- **Software** Win10
- SDR++: v1.1.0 Built at 20:19:27, Apr 10 2024

**Bug Description**
Spectrum display disappears after computer goes into nap mode (no user activity for some minutes => black screens on all attached monitors). After moving mouse, or touching a key, to wake the PC up from its nap, only the WF is shown in SDR++. I'm unable to find a "grab line" to move the border between spectrum and WF. Unchecking "Show Waterfall" in the display menu just swaps from waterfall to spectrum view but there is no way to show both. I never experienced this behaviour in earlier versions of SDR++.

**Steps To Reproduce**
1. ... See above..

Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).
After PC wake-up from nap:
[image]

After unchecking "Show Warerfall": 
[image]

AlexandreRouma, 2024-04-15:
Seems to have come from a bad commit pushed on april 10th.
Should now be fixed, let me know if not.

AlexandreRouma, 2024-04-15:
Additional note: If the broken FFT display persists after restarting SDR++, delete the `config.json` file to reset the main configuration.

closed 2024-04-15 by AlexandreRouma

JohBod, 2024-04-15:
Thank you for the quick fix! Will test immediately. Regarding the deletion of config.json, I figured out that trick myself ;)