AlexandreRouma/SDRPlusPlus #1382: Unusual (fixed) height issue with some Table components at High-DPI (Android)

boybook, 2024-04-15:
**Hardware**
- CPU: Qualcomm Snapdragon 8s Gen 3 Mobile Platform
- RAM: 16GB
- GPU: ~
- SDR: RTL-SDR

**Software**
- Operating System: Android 14
- SDR++: v1.1.0 (Built at 04:27:31, Apr 15 2024)

**Bug Description**
This is just a UI issue.
Unusual (fixed) height issue with some Table components at High-DPI (Android).

**Steps To Reproduce**
1. Open the Android version of SDR++ directly (or manually modify High-DPI Scaling in Display on other platforms.
2. As you can see, some of the Table components in the UI, such as the height of the tables in "Frequency Manager" and "Module Manager", are too short.

**Screenshots**
[image]

**Additional info**

https://github.com/boybook/SDRPlusPlus/commit/8f3c1992c127d4d4b264f753af1785129eee596b

```
[core/src/gui/menus/module_manager.cpp]
 
@@ -39,7 +39,7 @@ namespace module_manager_menu {
        ImVec2 btnSize = ImVec2(lheight, lheight - 1);
        ImVec2 textOff = ImVec2(3.0f * style::uiScale, -5.0f * style::uiScale);

        if (ImGui::BeginTable("Module Manager Table", 3, ImGuiTableFlags_Borders | ImGuiTableFlags_RowBg | ImGuiTableFlags_ScrollY, ImVec2(0, 200))) {
        if (ImGui::BeginTable("Module Manager Table", 3, ImGuiTableFlags_Borders | ImGuiTableFlags_RowBg | ImGuiTableFlags_ScrollY, ImVec2(0, 200 * style::uiScale))) {
            ImGui::TableSetupColumn("Name");
            ImGui::TableSetupColumn("Type");
            ImGui::TableSetupColumn("", ImGuiTableColumnFlags_WidthFixed, cellWidth);
```

```
[misc_modules/frequency_manager/src/main.cpp]
 
@@ -485,7 +485,7 @@ class FrequencyManagerModule : public ModuleManager::Instance {
        }

        // Bookmark list
        if (ImGui::BeginTable(("freq_manager_bkm_table" + _this->name).c_str(), 2, ImGuiTableFlags_Borders | ImGuiTableFlags_RowBg | ImGuiTableFlags_ScrollY, ImVec2(0, 200))) {
        if (ImGui::BeginTable(("freq_manager_bkm_table" + _this->name).c_str(), 2, ImGuiTableFlags_Borders | ImGuiTableFlags_RowBg | ImGuiTableFlags_ScrollY, ImVec2(0, 200.0f * style::uiScale))) {
            ImGui::TableSetupColumn("Name");
            ImGui::TableSetupColumn("Bookmark");
            ImGui::TableSetupScrollFreeze(2, 1);
```

AlexandreRouma, 2024-04-15:
Ok yeah should just be a matter of multiplying with the current UI scale. The one in the frequency manager has to be updated as well.

AlexandreRouma, 2024-04-16:
Not sure why this was closed

AlexandreRouma, 2024-04-16:
Fixed

closed 2024-04-16 by AlexandreRouma