AlexandreRouma/SDRPlusPlus #1388: Outdated RTL-SDR DLL Issue

lutusp, 2024-04-23:
The relatively new RTL-SDR V4 dongle works very well with SDR++, but only after some reconfigurations that shouldn't be necessary. In particular, on Windows -- where most SDR++ users live -- it's necessary to replace an outdated DLL with a current one that supports the most recent RTL-SDR dongle.

The current DLL that RTL-SDR V4 needs is located here:
    https://github.com/rtlsdrblog/rtl-sdr-blog/releases/download/1.3.5/Release.zip
The date on the current DLL is March 27, 2024.

The corresponding DLL bundled with SDR++ has a date of January 4, 2020 and won't work with the newer RTL-SDR versions.

I was planning to write an install/use article about SDR++ and the RTL-SDR V4, but to avoid burdening my readers with too much detail, I decided to test Windows first -- after all, that's the most common platform. But even on Windows, SDR++ won't detect or use the RTL-SDR V4 device without some DLL swapping.

Please consider bundling the current RTL-SDR DLL with your Windows release package, which BTW has a date of yesterday.

Yours is a first-rate application. Thanks for reading!

AlexandreRouma, 2024-04-23:
No, it's not as simple as using the rtlsdrblog version.

That fork has a hack meant for SDR# that breaks Bias-T control. This has been discussed with RTL-SDRblog themselves. Switching to it would break the bias-t for everyone, including V4 users.

The real solution, and what has been worked on for a while, is compiling the upstream Osmocom librtlsdr for windows which has V4 support but not the SDR# hack, but I don't have time for that stuff currently.

lutusp, 2024-04-23:
I should have known it wasn't as simple as I had imagined. Thanks for setting me straight. Closing this issue.

WB0OCV, 2024-04-23:
I know there is an alternative set of dlls provided by Sultan-papgani that allow for additional AGC and gain settins for the RTL-SDR Blog V4.  Those have worked with past versions of SDR++, but with the latest release on 20240422, the modified dll (new_rtl_sdr_source.dll) causes the program to hang when launching the program in a folder from a prior installation.  If one deletes the json files, the program will start, but the new_rtl_sdr module isn't recognized.  If one tires to create the new module, the program crashes.  I just wanted to make you aware of this issue.  Thank you for such a wonderful program.  
Ken -- WBØOCV

AlexandreRouma, 2024-04-23:
Report this to the module author, there is nothing I can do about it.

And side note, I really don't recommend using that thing, all the controls exposed are already set to sane values automatically by librtlsdr. Setting them to anything else is guaranteeing a worse reception.

Sultan-papagani, 2024-04-24:
my rtl-sdr [source](https://github.com/Sultan-papagani/sdrpp_rtlsdr_source) is fixed right now. b.t.w i get better reception with the filter & seperate vga,lna,mixer gains.

AlexandreRouma, 2024-05-26:
I'll set this as a feature request for switching to the recent osmocomm drivers.

closed 2024-10-03 by AlexandreRouma