AlexandreRouma/SDRPlusPlus #1398: No PlutoSDR source in MacOS 14

denisbondar, 2024-05-05:
**Hardware**
- CPU: Apple M1 Pro
- RAM: 16 ГБ
- GPU: Apple M1 Pro (Metal 3)
- SDR: LibreSDR (PlutoSDR) network and USD connection

**Software**
- Operating System: macOS 14.4.1 
- SDR++: v1.2.0 (Built at 14:40:56, May 4 2024)

**Bug Description**
PlutoSDR is missing from the list of signal sources.

**Screenshots**
[image]
[image]
[image]
[image]

**Additional info**
I deleted the `~/Library/Application Support/sdrpp/config.json`. I also deleted the entire `~/Library/Application Support/sdrpp` directory. This did not give any result.

The libiio library is installed. Other applications like GQRX work with both USB and network connection.

AlexandreRouma, 2024-05-05:
Show startup logs.

denisbondar, 2024-05-05:
Can you please tell me where I can find this log file?

AlexandreRouma, 2024-05-05:
There is no log file. Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

denisbondar, 2024-05-06:
**Unknown argument**
```text
$ ./Applications/SDR++.app/Contents/MacOS/sdrpp -c
[06/05/2024 09:16:11.000] [INFO] SDR++ v1.2.0
Unknown argument
-a --addr		Server mode address
   --autostart		Automatically start the SDR after loading
-h --help		Show help
-p --port		Server mode port
-r --root		Root directory, where all config files are stored
-s --server		Run in server mode
```

denisbondar, 2024-05-06:
Without `-c`:
```text
$ ./Applications/SDR++.app/Contents/MacOS/sdrpp                                                                 255 ↵
[06/05/2024 09:18:01.000] [INFO] SDR++ v1.2.0
[06/05/2024 09:18:01.000] [INFO] Loading config
[06/05/2024 09:18:01.000] [WARN] ConfigManager locked, waiting...
[06/05/2024 09:18:01.000] [ERROR] Glfw Error 65548: Cocoa: Regular windows do not have icons on macOS
[06/05/2024 09:18:02.000] [INFO] Loading icons
[06/05/2024 09:18:02.000] [INFO] Loading band plans
[06/05/2024 09:18:02.000] [INFO] Loading band plans color table
[06/05/2024 09:18:02.000] [ERROR] Menu element is missing name key
[06/05/2024 09:18:02.000] [INFO] Loading modules
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/rtl_sdr_source.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/meteor_demodulator.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/airspy_source.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/network_sink.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/scanner.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib
[06/05/2024 09:18:02.000] [ERROR] Couldn't load /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib: dlopen(/Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib, 0x0005): Library not loaded: @rpath/libxml2.2.dylib
  Referenced from: <E731C8AF-AE3B-3AF9-B773-FD7FBC946263> /Users/denisbondar/Applications/SDR++.app/Contents/Frameworks/iio
  Reason: tried: '/Users/denisbondar/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libxml2.2.dylib' (no such file), '/Users/denisbondar/Applications/SDR++.app/Contents/Plugins/../Frameworks/libxml2.2.dylib' (no such file), '/Users/denisbondar/Applications/SDR++.app/Contents/Plugins/../Frameworks/libxml2.2.dylib' (no such file), '/Users/denisbondar/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libxml2.2.dylib' (no such file), '/Users/denisbondar/Applications/SDR++.app/Contents/MacOS/../Frameworks/libxml2.2.dylib' (no such file)
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib
[06/05/2024 09:18:02.000] [ERROR] Couldn't load /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib: dlopen(/Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib, 0x0005): Library not loaded: libsdrplay_api.so.3.14
  Referenced from: <93052A15-FD52-30E6-8B23-5961065EBE27> /Users/denisbondar/Applications/SDR++.app/Contents/Plugins/sdrplay_source.dylib
  Reason: tried: 'libsdrplay_api.so.3.14' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibsdrplay_api.so.3.14' (no such file), 'libsdrplay_api.so.3.14' (no such file), '/Users/denisbondar/Applications/SDR++.app/Contents/Resources/libsdrplay_api.so.3.14' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/denisbondar/Applications/SDR++.app/Contents/Resources/libsdrplay_api.so.3.14' (no such file), '/Users/denisbondar/Applications/SDR++.app/Contents/Resources/libsdrplay_api.so.3.14' (no such file)
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/radio.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/rigctl_server.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/audio_sink.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/frequency_manager.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/recorder.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/file_source.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/bladerf_source.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/airspyhf_source.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/spyserver_source.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/rigctl_client.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/new_portaudio_sink.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/hermes_source.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/limesdr_source.dylib
[06/05/2024 09:18:02.000] [INFO] Loading /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/discord_integration.dylib
[... cut ...]
[06/05/2024 09:18:03.000] [INFO] Running post-init for Recorder
[06/05/2024 09:18:03.000] [INFO] Running post-init for Rigctl Server
[06/05/2024 09:18:03.000] [INFO] Running post-init for SDR++ Server Source
[06/05/2024 09:18:03.000] [INFO] Running post-init for SpyServer Source
[06/05/2024 09:18:03.000] [INFO] Ready.
2024-05-06 09:18:11.180 sdrpp[16961:1179352] TSM AdjustCapsLockLEDForKeyTransitionHandling - _ISSetPhysicalKeyboardCapsLockLED Inhibit
[06/05/2024 09:18:12.000] [INFO] Exiting successfully
```

denisbondar, 2024-05-06:
It should be noted that I have libxml2 installed using brew.
```
$ ls /opt/homebrew/Cellar/libxml2/2.12.6/lib/
cmake           libxml2.2.dylib libxml2.dylib   pkgconfig       python3.11      python3.12
```
However, judging by the log, the library file should be inside app.

AlexandreRouma, 2024-05-06:
no libxml2 is supposed to come with every MacOS install.

denisbondar, 2024-05-06:
How can I solve this problem on my side?

closed 2024-10-03 by AlexandreRouma

g-sed, 2025-01-11:
Hi Alexandre, I have the same problem, please reopen this issue.

**Hardware**
CPU: Apple M3 Max
RAM: 128 GB
SDR: ADALM-PLUTO v0.39
**Software**
Operating System: macOS 14.6.1
SDR++: v1.2.1
**Bug Description**
PlutoSDR is missing from the list of signal sources.

--------------- snip -----------------

./sdrpp   
[11/01/2025 05:30:00.000] [INFO] SDR++ v1.2.1
[11/01/2025 05:30:00.000] [INFO] Loading config
[11/01/2025 05:30:00.000] [WARN] ConfigManager locked, waiting...
[11/01/2025 05:30:00.000] [ERROR] Glfw Error 65548: Cocoa: Regular windows do not have icons on macOS
[11/01/2025 05:30:00.000] [INFO] Loading icons
[11/01/2025 05:30:00.000] [INFO] Loading band plans
[11/01/2025 05:30:00.000] [INFO] Loading band plans color table
[11/01/2025 05:30:00.000] [INFO] Loading modules
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rtl_sdr_source.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/meteor_demodulator.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/airspy_source.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/network_sink.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/scanner.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib
[11/01/2025 05:30:00.000] [ERROR] Couldn't load /Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib: dlopen(/Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib, 0x0005): Library not loaded: @rpath/libxml2.2.dylib
  Referenced from: <5A6DC127-0ACF-35AD-9DA9-A78DC3C4CDB6> /Applications/SDR++.app/Contents/Frameworks/iio
  Reason: tried: '/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libxml2.2.dylib' (no such file), '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libxml2.2.dylib' (no such file), '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libxml2.2.dylib' (no such file), '/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libxml2.2.dylib' (no such file), '/Applications/SDR++.app/Contents/MacOS/../Frameworks/libxml2.2.dylib' (no such file)
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib
[11/01/2025 05:30:00.000] [ERROR] Couldn't load /Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib: dlopen(/Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib, 0x0005): Library not loaded: @rpath/libsdrplay_api.so.3
  Referenced from: <D34FE7BB-673E-327F-AA00-461A169A1DA2> /Applications/SDR++.app/Contents/Plugins/sdrplay_source.dylib
  Reason: tried: '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/MacOS/../Frameworks/libsdrplay_api.so.3' (no such file)
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/radio.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rigctl_server.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/audio_sink.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/frequency_manager.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/recorder.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/file_source.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/network_source.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/bladerf_source.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/airspyhf_source.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rfnm_source.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/spyserver_source.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/fobossdr_source.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rigctl_client.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/new_portaudio_sink.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/hermes_source.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/limesdr_source.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/discord_integration.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rfspace_source.dylib
[11/01/2025 05:30:00.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rtl_tcp_source.dylib
[... cut ...]
[11/01/2025 05:30:01.000] [INFO] Running post-init for RTL-SDR Source
[11/01/2025 05:30:01.000] [INFO] Running post-init for RTL-TCP Source
[11/01/2025 05:30:01.000] [INFO] Running post-init for Radio
[11/01/2025 05:30:01.000] [INFO] Running post-init for Recorder
[11/01/2025 05:30:01.000] [INFO] Running post-init for Rigctl Server
[11/01/2025 05:30:01.000] [INFO] Running post-init for SDR++ Server Source
[11/01/2025 05:30:01.000] [INFO] Running post-init for SpyServer Source
[11/01/2025 05:30:01.000] [INFO] Ready.

MoonFingerRF, 2025-01-11:
This is definitely a problem with SDR++, I too am having the same problem

M4 Max MacBook pro 36GB
OS: Sequoia 15.2

Hamgeek adi pluto+

PlutoSDR not showing in sources but plutosdr_souce.dylib shows up in the plugins folder. Unable to add it to sources.

Log shows an error loading the plutosdr plugin specifically.

log:
[11/01/2025 18:24:47.000] [INFO] SDR++ v1.2.1
[11/01/2025 18:24:47.000] [INFO] Loading config
[11/01/2025 18:24:47.000] [WARN] ConfigManager locked, waiting...
[11/01/2025 18:24:48.000] [ERROR] Glfw Error 65548: Cocoa: Regular windows do not have icons on macOS
[11/01/2025 18:24:48.000] [INFO] Loading icons
[11/01/2025 18:24:48.000] [INFO] Loading band plans
[11/01/2025 18:24:48.000] [INFO] Loading band plans color table
[11/01/2025 18:24:48.000] [INFO] Loading modules
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rtl_sdr_source.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/meteor_demodulator.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/airspy_source.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/network_sink.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/scanner.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib
[11/01/2025 18:24:48.000] [ERROR] Couldn't load /Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib: dlopen(/Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib, 0x0005): Library not loaded: @rpath/libxml2.2.dylib
  Referenced from: <5A6DC127-0ACF-35AD-9DA9-A78DC3C4CDB6> /Applications/SDR++.app/Contents/Frameworks/iio
  Reason: tried: '/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libxml2.2.dylib' (no such file), '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libxml2.2.dylib' (no such file), '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libxml2.2.dylib' (no such file), '/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libxml2.2.dylib' (no such file), '/Applications/SDR++.app/Contents/MacOS/../Frameworks/libxml2.2.dylib' (no such file)
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib
[11/01/2025 18:24:48.000] [ERROR] Couldn't load /Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib: dlopen(/Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib, 0x0005): Library not loaded: @rpath/libsdrplay_api.so.3
  Referenced from: <D34FE7BB-673E-327F-AA00-461A169A1DA2> /Applications/SDR++.app/Contents/Plugins/sdrplay_source.dylib
  Reason: tried: '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/MacOS/../Frameworks/libsdrplay_api.so.3' (no such file)
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/radio.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rigctl_server.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/audio_sink.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/frequency_manager.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/recorder.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/file_source.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/network_source.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/bladerf_source.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/airspyhf_source.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rfnm_source.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/spyserver_source.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/fobossdr_source.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rigctl_client.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/new_portaudio_sink.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/hermes_source.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/limesdr_source.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/discord_integration.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rfspace_source.dylib
[11/01/2025 18:24:48.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rtl_tcp_source.dylib
[... cut ...]
[11/01/2025 18:24:48.000] [INFO] Running post-init for Recorder
[11/01/2025 18:24:48.000] [INFO] Running post-init for Rigctl Server
[11/01/2025 18:24:48.000] [INFO] Running post-init for SDR++ Server Source
[11/01/2025 18:24:48.000] [INFO] Running post-init for SpyServer Source
[11/01/2025 18:24:48.000] [INFO] Ready.
2025-01-11 18:24:48.634 sdrpp[32873:3795700] +[IMKClient subclass]: chose IMKClient_Modern
2025-01-11 18:24:48.634 sdrpp[32873:3795700] +[IMKInputSession subclass]: chose IMKInputSession_Modern

MoonFingerRF, 2025-01-12:
Just got it to work!

I installed libxml2 with homebrew(make sure pkg-config is installed first):

`brew install pkg-config
brew install libxml2`

Then copied the libxml2.2.dylib file from its location in homebrew to the frameworks folder for sdr++.

/opt/homebrew/Cellar/libxml2/2.13.5/lib/libxml2.2.dylib
to
/path/to/SDR++.app/Contents/Frameworks/

AlexandreRouma, 2025-01-12:
libxml2 is provided by the system and must not be copied to the application.

MoonFingerRF, 2025-01-12:
According to the log file it is not checking the system folder for libxml2 and only looking in the frameworks folder. I'm not sure why it's not finding it by default but this was the only way I was able to get this to work on my machine

denisbondar, 2025-01-12:
> I installed libxml2 with homebrew(make sure pkg-config is installed first):
> Then copied the libxml2.2.dylib file from its location in homebrew to the frameworks folder for sdr++.

As a one-off way to solve a problem - maybe that's fine. But I would like the author to solve this problem on the SDR++ side.
Instead, the author refuses to acknowledge the problem, which is very frustrating.

g-sed, 2025-01-12:
Hi MoonfingerRF, I tried your workaround without success. SDR++ crashes during start. 

Any ideas?

----------- snip -----------

./sdrpp                                                                                                    
[12/01/2025 11:09:51.000] [INFO] SDR++ v1.2.1
[12/01/2025 11:09:51.000] [INFO] Loading config
[12/01/2025 11:09:51.000] [WARN] ConfigManager locked, waiting...
[12/01/2025 11:09:51.000] [ERROR] Glfw Error 65548: Cocoa: Regular windows do not have icons on macOS
[12/01/2025 11:09:51.000] [INFO] Loading icons
[12/01/2025 11:09:51.000] [INFO] Loading band plans
[12/01/2025 11:09:51.000] [INFO] Loading band plans color table
[12/01/2025 11:09:51.000] [INFO] Loading modules
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rtl_sdr_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/meteor_demodulator.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/airspy_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/network_sink.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/scanner.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib
[12/01/2025 11:09:51.000] [ERROR] Couldn't load /Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib: dlopen(/Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib, 0x0005): Library not loaded: @rpath/libsdrplay_api.so.3
  Referenced from: <D34FE7BB-673E-327F-AA00-461A169A1DA2> /Applications/SDR++.app/Contents/Plugins/sdrplay_source.dylib
  Reason: tried: '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/MacOS/../Frameworks/libsdrplay_api.so.3' (no such file)
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/radio.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rigctl_server.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/audio_sink.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/frequency_manager.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/recorder.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/file_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/network_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/bladerf_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/airspyhf_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rfnm_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/spyserver_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/fobossdr_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rigctl_client.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/new_portaudio_sink.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/hermes_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/limesdr_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/discord_integration.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rfspace_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/rtl_tcp_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/hackrf_source.dylib
[12/01/2025 11:09:51.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/sdrpp_server_source.dylib
[12/01/2025 11:09:51.000] [INFO] Initializing Airspy Source (airspy_source)
[12/01/2025 11:09:51.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[12/01/2025 11:09:51.000] [INFO] Initializing Audio Sink (audio_sink)
[12/01/2025 11:09:51.000] [INFO] Initializing Audio Source (audio_source)
[12/01/2025 11:09:51.000] [ERROR] Module 'audio_source' doesn't exist
[12/01/2025 11:09:51.000] [INFO] Initializing BladeRF Source (bladerf_source)
[12/01/2025 11:09:51.000] [ERROR] Could not list devices -7
[12/01/2025 11:09:51.000] [INFO] Initializing File Source (file_source)
[12/01/2025 11:09:51.000] [INFO] Initializing FobosSDR Source (fobossdr_source)
[12/01/2025 11:09:51.000] [INFO] Initializing Frequency Manager (frequency_manager)
[... cut ...]
[12/01/2025 11:09:51.000] [INFO] Initializing Network Sink (network_sink)
[12/01/2025 11:09:51.000] [INFO] Initializing Network Source (network_source)
[12/01/2025 11:09:51.000] [INFO] Initializing PerseusSDR Source (perseus_source)
[12/01/2025 11:09:51.000] [ERROR] Module 'perseus_source' doesn't exist
[12/01/2025 11:09:51.000] [INFO] Initializing PlutoSDR Source (plutosdr_source)
libc++abi: terminating due to uncaught exception of type std::runtime_error: Name already exists
zsh: abort      ./sdrpp

denisbondar, 2025-01-12:
> libxml2 is provided by the system and must not be copied to the application.

That makes perfect sense!
However, it doesn't work. I have described the reason for this [here](https://github.com/AlexandreRouma/SDRPlusPlus/issues/1398#issuecomment-2095273260).

```
06/05/2024 09:18:02.000] [ERROR] Couldn't load /Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib:
dlopen(/Users/denisbondar/Applications/SDR++.app/Contents/Resources/../Plugins/plutosdr_source.dylib, 0x0005):
Library not loaded: @rpath/libxml2.2.dylib

  Referenced from: <E731C8AF-AE3B-3AF9-B773-FD7FBC946263> /Users/denisbondar/Applications/SDR++.app/Contents/Frameworks/iio

  Reason: tried:
'/Users/denisbondar/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libxml2.2.dylib' (no such file),
'/Users/denisbondar/Applications/SDR++.app/Contents/Plugins/../Frameworks/libxml2.2.dylib' (no such file),
'/Users/denisbondar/Applications/SDR++.app/Contents/Plugins/../Frameworks/libxml2.2.dylib' (no such file),
'/Users/denisbondar/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libxml2.2.dylib' (no such file),
'/Users/denisbondar/Applications/SDR++.app/Contents/MacOS/../Frameworks/libxml2.2.dylib' (no such file)
```

The search is performed inside SDR++.app/Contents.

sarumjanuch, 2025-01-29:
+1 here. And yes, libxml2 is resolved only from frameworks folder, and not from the system path. Putting copy of libxml2 into frameworks folder does fix this issue.

denisbondar, 2025-02-06:
@AlexandreRouma Please consider revisiting this issue. Don't force us to give up your wonderful tool in favor of others.

sarumjanuch, 2025-02-06:
I have faced another issue, that I had my Pluto source duplicated for ethernet connection and that was crashing sdrpp, anyone else had that? I have a fix for it, just not sure should i create PR.. or it's only me?

denisbondar, 2025-02-06:
By the way. To make SDR++ work temporarily you don't need to copy the library file at all. It is enough to create a symbolic link as follows:
```
sudo ln -s /opt/homebrew/opt/libxml2/lib/libxml2.2.dylib /Applications/SDR++.app/Contents/Frameworks/libxml2.2.dylib
```

matteocordray, 2025-02-18:
+1 on this being an issue on an Intel Mac for the Pluto Plus (over network).

Did the same steps as @MoonFingerRF to copy the libxml2 `dylib` file and worked great. For Intel Mac:

```bash
brew install pkg-config
brew install libxml2

cp usr/local/Cellar/libxml2/2.13.5/lib/libxml2.2.dylib /Applications/SDR++.app/Contents/Frameworks
```

Then run sdrpp as normal, plutoSDR shows up as a valid source.

bangpradyumna, 2025-02-18:
In my macbook pro M4, as soon as I make the symbolic link, SDR++ stops opening

denisbondar, 2025-02-18:
> In my macbook pro M4, as soon as I make the symbolic link, SDR++ stops opening

Run sdrpp from terminal:
```
$ /Applications/SDR++.app/Contents/MacOS/sdrpp
```
You'll see the log and probably an error in it.

alexey-lysiuk, 2025-03-12:
According to provided error message, `libxml2.2.dylib` is loaded by `SDR++.app/Contents/Frameworks/iio`.
Output the following console command reveals that it references via runtime path.

```
otool -L SDR++.app/Contents/Frameworks/iio | fgrep libxml
```
```
@rpath/libxml2.2.dylib (compatibility version 10.0.0, current version 10.9.0)
```

The next console command shows to check runtime paths.

```
otool -l SDR++.app/Contents/Frameworks/iio
```
There is only one entry.
```
Load command 19
          cmd LC_RPATH
      cmdsize 40
         path @loader_path/../Frameworks (offset 12)
```

Apparently, `libxml2.2.dylib` is missing from the `SDR++.app/Contents/Frameworks`.
To use system's `libxml2.2.dylib`, it must be referenced via absolute path, i.e., `/usr/lib/libxml2.2.dylib`.

denisbondar, 2025-03-12:
> Apparently, libxml2.2.dylib is missing from the SDR++.app/Contents/Frameworks.

That's the way it is.
libxml2.2.dylib is missing inside SDR++.app. Also SDR++.app tries to find it in several directories that are visible when launching SDR++.app from CLI.
Mentioned this above in a [comment](https://github.com/AlexandreRouma/SDRPlusPlus/issues/1398#issuecomment-2585691894).

Unfortunately, the author refuses to address this error.

alexey-lysiuk, 2025-03-12:
Curiously enough, system's libXML is found during GitHub Actions CI builds, at [Install libiio step](https://github.com/alexey-lysiuk/SDRPlusPlus/blob/64315ebc618ffab9293370e22dc69793322b7ee4/.github/workflows/build_all.yml#L118-L119).
```
-- Found LibXml2: /Applications/Xcode_15.2.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.2.sdk/usr/lib/libxml2.tbd (found version "2.9.13")
```
```
-- Found LibXml2: /Applications/Xcode_15.4.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.5.sdk/usr/lib/libxml2.tbd (found version "2.9.13")
```
However, it links with other .dylib, picked from Homebrew probably.

AlexandreRouma, 2025-03-12:
> Unfortunately, the author refuses to address this error.

"Refuses"?

Buy me a modern Mac then.... I already said it works fine on my old Catalina mac (2009 mac pro). If I can't replicate your issue I can't solve shit.

> However, it links with other .dylib, picked from Homebrew probably.

On slightly older MacOS versions (like a year or so) it works file, it has worked for fine years. Apple changed something that makes xcode link stuff wrong and I can't debug it. The only things people have suggested as "fixes" right now are hacks, not actual fixes.

denisbondar, 2025-03-12:
> On slightly older MacOS versions (like a year or so) it works fine, it has worked for fine years. Apple changed something that makes xcode link stuff wrong and I can't debug it. The only things people have suggested as "fixes" right now are hacks, not actual fixes.

Is it not possible to use the system libxml2.2 instead of trying to load this library from the SDR++.app package.

AlexandreRouma, 2025-03-12:
> Is it not possible to use the system libxml2.2 instead of trying to load this library from the SDR++.app package.

It used to. libxml2 has never been included in any SDR++ bundle ever released.

denisbondar, 2025-03-12:
> It used to. libxml2 has never been included in any SDR++ bundle ever released.

Then I'm probably missing something. Because what I see [in the log](https://github.com/AlexandreRouma/SDRPlusPlus/issues/1398#issuecomment-2585691894) is that the file is searched in specific directories within the package.

alexey-lysiuk, 2025-03-13:
It's fixed in aa2b4b1c5814cc2f832898a9e4a1bdfc38e7ac8d. Note that nightly build was not updated because of failed Windows job.