AlexandreRouma/SDRPlusPlus #1399: Pluto-SDR IP-Address Android

dionysos83, 2024-05-06:
Not possible changing IP-Address of Adalm Pluto in Android build. 

Hardware Xiaomi Mi9t pro - stock android 11
MiUi 12.5.1

AlexandreRouma, 2024-10-03:
Not a bug, just not an implemented feature. It uses the standard IP address of any plutosdr. It will not be compatible with one that has been configured differently. I'll leave this open as a feature request instead.

Yura1993, 2025-01-08:
> Not a bug, just not an implemented feature. It uses the standard IP address of any plutosdr. It will not be compatible with one that has been configured differently. I'll leave this open as a feature request instead.

Please add manual adding of PlutoSDR ip adress, like on SDR#
And we need also support of PuzhiSDR P203 mini (chinese replic of Pluto), its works on SDR# fine and on SDR++ its didn't detect it at all