AlexandreRouma/SDRPlusPlus #1400: Support for SDRplay_RSP_API-Linux-3.15.1

M4rc-Andre, 2024-05-10:
With this new API from SDRplay for the RSPdx-R2 SDR++ does not recognize any SDRplay receivers under Ubuntu Linux. Had to go back to 3.14. Please customize SDR++ for this new API.

nmaster2042, 2024-05-12:
I think things will be like for adding support for RSP1B, see here: 

https://github.com/AlexandreRouma/SDRPlusPlus/issues/1311

After searching the new API documentation, RSPDs-R2 is a reworked RSPDX, so same calls then this one from the APi point of view. Exactly like RSP1B uses RSP1A specific API calls.

This is also confirmed reading support for these new devices in the SoapySDR driver.

alphafox02, 2024-05-12:
Did you build from source against the new api with it installed or did use a deb built against 3.14 then install the 3.15 api? If its the latter I would expect the behavior you’re seeing.

nmaster2042, 2024-05-13:
First it would be necessary to add support for the DXR2 in the sdrplay_source same way it has been done for RSP1B, then it must be built against the new 3.15 API to work.

Personally I own the RSP1B, and this worked this way, and it should work same way for DXR2 as I explained in my previous message.

I don't own the RSP DXR2.

You have to build from source after modify sdrplay_source itself.

nmaster2042, 2024-05-13:
To confirm what I think about adding support for new RSPs (DXR2), on thwitter SDRPlay put this:

[image]

AlexandreRouma, 2024-05-28:
The night builds have been updated to use the 3.15 API, for device support follow #1311 instead.

closed 2024-05-28 by AlexandreRouma