AlexandreRouma/SDRPlusPlus #1403: No more RTL-SDR source module in SDR++

VinCiv, 2024-05-20:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel(R) Celeron(R) N4020 CPU @ 1.10GHz
- RAM: 4 GB
- GPU: 
- SDR: RTL-SDR NOOELEC 

**Software**
- Operating System: Ubuntu 24.04 x64
- SDR++: 1.2.0 - May 15 2024

**Bug Description**
The RTL-SDR source does not appear in the available sources in the main combo-box of the Source sub-menu, nor in the Module Manager.

**Steps To Reproduce**
1. Just unfold main combo of Source sub-menu: it lacks RTL-SDR

**Screenshots**
Added

**Additional info**
No more to say

[image]

AlexandreRouma, 2024-05-20:
Start sdr++ from a terminal and show all startup logs.

Void0111, 2024-06-05:
I'm facing these problems too and my Terminal logs are pretty much the same.

KubaPro010, 2024-06-05:
Run `sudo apt install rtlsdr`

Void0111, 2024-06-05:
There is no rtlsdr apt Package, I tried rtl-sdr. But still the same problem

KubaPro010, 2024-06-05:
If it says `Couldn't load /usr/lib/sdrpp/plugins/rtl_sdr_source.so: librtlsdr.so.0: cannot open shared object file: No such file or directory` then that means that librtlsdr is not installed, rtl-sdr should install it, try installing `librtlsdr`

AlexandreRouma, 2024-06-05:
I've already explained it in other places, but osmocomm changed the soname version to 2. NO APP will work with it, you can try gqrx, cubic, etc, it won't load the newer driver without being recompiled. There is NOTHING I can do.

Your only option is to wait for your distro to ship with rtlsdr2 by default or recompile all your SDR apps yourself. I raised the issue to osmocomm serveral months ago and nothing was done about it.

closed 2024-06-05 by AlexandreRouma