AlexandreRouma/SDRPlusPlus #1405: Not working on LG V20

yovko82, 2024-05-22:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: SnapDragon 820
- RAM:4Gb
- GPU: Ardeno 530
- SDR: (Remote or local? If remote, what protocol?) LOCAL

**Software**
Android 8, Oreo
- SDR++: 1.2.0

**Bug Description**
Don`t support V4. I don`t see it to work

**Steps To Reproduce**
1. ...Install
2. ...Run
3. ...Add RTL-SDR and don`t work

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

yovko82, 2024-05-22:
SDRpp is not installing either. It gaves error parsing and that`s it.

KubaPro010, 2024-05-22:
RTL SDR V4 isn't supported by SDR++

KubaPro010, 2024-05-22:
Yes, infact i asked ChatGPT to anylaze the build.gradle file, here's what ChatGPT had to say:

> `The build.gradle script you provided specifies that the minimum SDK version (minSdkVersion) is set to 28, which corresponds to Android 9.0 (Pie). This configuration means that the application will only run on devices with Android 9.0 or higher.`

AlexandreRouma, 2024-05-22:
@KubaPro010 Complete bullshit, the RTL-SDRv4 is supported on android.

@yovko82 Your phone is too old. Android 9 is necessary.

closed 2024-05-22 by AlexandreRouma

KubaPro010, 2024-05-22:
.I didn't mean on Android but in general, i guess thanks for correcting me.

AlexandreRouma, 2024-05-22:
the V4 is supported everywhere. Just that on windows you need to replace a DLL which is explained on the manufacturer's website.

yovko82, 2024-05-22:
No problem. I already try it on Android 14 and it`s working perfect with RTL-SDR v4