AlexandreRouma/SDRPlusPlus #1406: Rfspace Module Crashes or does not respond

zekiispa, 2024-05-25:
**Hardware**
- CPU: i913900k
- RAM: 32gb
- GPU: rtx4080
- SDR: Local

**Software**
Winsows11 23H2

**Bug Description**
After connecting to the device, it does not respond or crashes and shuts down when started.

**Steps To Reproduce**
1. Connect
2. .Run

C:\Program Files\SDR\sdrpp_windows_x64>sdrpp.exe -c
[25/05/2024 08:36:29.000] [INFO] SDR++ v1.2.0
[25/05/2024 08:36:29.000] [INFO] Loading config
[25/05/2024 08:36:29.000] [INFO] Using OpenGL 3.0
[25/05/2024 08:36:29.000] [INFO] Loading icons
[25/05/2024 08:36:29.000] [INFO] Loading band plans
[25/05/2024 08:36:29.000] [INFO] Loading band plans color table
[25/05/2024 08:36:29.000] [INFO] Loading modules
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/airspyhf_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/airspy_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/audio_sink.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/audio_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/bladerf_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/discord_integration.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/file_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/frequency_manager.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/hackrf_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/hermes_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/iq_exporter.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/limesdr_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/m17_decoder.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/meteor_demodulator.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/network_sink.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/perseus_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/plutosdr_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/radio.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/recorder.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/rfspace_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/rigctl_client.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/rigctl_server.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/rtl_sdr_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/rtl_tcp_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/scanner.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/sdrplay_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/sdrpp_server_source.dll
[25/05/2024 08:36:29.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/spyserver_source.dll
[25/05/2024 08:36:29.000] [INFO] Initializing Airspy Source (airspy_source)
[25/05/2024 08:36:29.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[25/05/2024 08:36:29.000] [INFO] Initializing Audio Sink (audio_sink)
[25/05/2024 08:36:29.000] [INFO] Initializing Audio Source (audio_source)
[25/05/2024 08:36:29.000] [INFO] New DSP samplerate: 48000.000000 (source samplerate is 48000.000000)
[25/05/2024 08:36:29.000] [INFO] Initializing BladeRF Source (bladerf_source)
[25/05/2024 08:36:29.000] [ERROR] Could not list devices -7
[25/05/2024 08:36:29.000] [INFO] Initializing File Source (file_source)
[25/05/2024 08:36:29.000] [INFO] Initializing Frequency Manager (frequency_manager)
[25/05/2024 08:36:29.000] [INFO] Initializing HackRF Source (hackrf_source)
[25/05/2024 08:36:29.000] [INFO] Initializing Hermes Source (hermes_source)
[25/05/2024 08:36:29.000] [INFO] Initializing LimeSDR Source (limesdr_source)
[25/05/2024 08:36:29.000] [INFO] Initializing Network Sink (network_sink)
[25/05/2024 08:36:29.000] [INFO] Initializing PerseusSDR Source (perseus_source)
perseus: perseus_init()
perseus: EXE: [C:\Program Files\SDR\sdrpp_windows_x64\sdrpp.exe]
perseus: path: [C:\Program Files\SDR\sdrpp_windows_x64\] name: [sdrpp.exe]
perseus: Found device with VID/PID 1E71:300C on BUS1 ADDR15
[... 371 lines cut ...]
[25/05/2024 08:37:01.000] [INFO] New DSP samplerate: 1228800.000000 (source samplerate is 1228800.000000)
[25/05/2024 08:37:01.000] [INFO] RFSpaceSourceModule 'RFspace Source': Menu Select!
[Resamp] predec: 4, interp: 625, decim: 768, inacc: 0.000000%, taps: 58368
[25/05/2024 08:37:02.000] [INFO] New DSP samplerate: 1228800.000000 (source samplerate is 1228800.000000)
[25/05/2024 08:37:02.000] [WARN] End
[25/05/2024 08:37:04.000] [INFO] RFSpaceSourceModule 'RFspace Source': Start!
[25/05/2024 08:37:04.000] [INFO] RFSpaceSourceModule 'RFspace Source': Tune: 100000000.000000!
[25/05/2024 08:37:08.000] [INFO] RFSpaceSourceModule 'RFspace Source': Stop!
[25/05/2024 08:37:10.000] [INFO] RFSpaceSourceModule 'RFspace Source': Start!
[25/05/2024 08:37:10.000] [INFO] RFSpaceSourceModule 'RFspace Source': Tune: 100000000.000000!
[25/05/2024 08:37:17.000] [INFO] RFSpaceSourceModule 'RFspace Source': Stop!
[Resamp] predec: 4, interp: 625, decim: 768, inacc: 0.000000%, taps: 58368
[25/05/2024 08:37:19.000] [INFO] New DSP samplerate: 1228800.000000 (source samplerate is 1228800.000000)

https://github.com/AlexandreRouma/SDRPlusPlus/assets/5446930/d1995bfe-caa6-4c90-a69d-c9faba1d409c

https://github.com/AlexandreRouma/SDRPlusPlus/assets/5446930/b6c37cc2-af09-423c-9752-d8c961916dda

[25/05/2024 08:37:19.000] [WARN] End
[25/05/2024 08:37:20.000] [INFO] RFSpaceSourceModule 'RFspace Source': Start!
[25/05/2024 08:37:20.000] [INFO] RFSpaceSourceModule 'RFspace Source': Tune: 100000000.000000!
[25/05/2024 08:37:26.000] [INFO] RFSpaceSourceModule 'RFspace Source': Stop!
[25/05/2024 08:37:28.000] [INFO] Exiting successfully

C:\Program Files\SDR\sdrpp_windows_x64>

zekiispa, 2024-05-25:
https://github.com/AlexandreRouma/SDRPlusPlus/assets/5446930/c01960fc-dbf9-4021-ae6f-19c52bd2b75e

AlexandreRouma, 2024-05-25:
You haven't said what device you're using

zekiispa, 2024-05-25:
Good morning,

Sory my bad RFspace CloudSDR http://www.rfspace.com/RFSPACE/CloudSDR.html

AlexandreRouma, 2024-05-26:
I can't replicate here on my CloudIQ, i'll ask Pieter from RFspace if he's able to replicate.

zekiispa, 2024-05-26:
Hello AlexandreRouma,

I have reactivated my old computer to redo my checks. The RF Space Cloud SDR worked without any issues. The USRP source DLL I compiled for the USRP also worked fine. (By the way, I learned how to compile the source code :) ) ... I think there is a problem with the USB and network on the motherboard of my new computer. I am sorry for the trouble and inconvenience caused.

https://github.com/AlexandreRouma/SDRPlusPlus/assets/5446930/34872265-0cd0-42dc-9287-2ff41ae47797

AlexandreRouma, 2024-05-26:
A network issue is highly unlikely to cause a crash like this. It should have timed out instead. I still think there is something wrong here.

zekiispa, 2024-05-27:
Hi, I think you're right,
I'm experimenting in detail. I'll send you the video and log files.

zekiispa, 2024-05-28:
Hello,
I reinstalled windows11. the result is the same, I only have problems with cloud sdr. USRP, Airspay, Hack rf, pluto etc... work fine. I can also use Cloud SDR this way. In the end, it works stably after a few tries :) 

https://github.com/AlexandreRouma/SDRPlusPlus/assets/5446930/1d7adf80-2751-468f-a654-3070d3b8df55

Maybe the video and logs will help you.

Best Regards
Zeki

C:\Program Files\SDR\sdrpp_windows_x64>sdrpp.exe -c
[29/05/2024 01:19:40.000] [INFO] SDR++ v1.2.0
[29/05/2024 01:19:40.000] [INFO] Loading config
[29/05/2024 01:19:40.000] [INFO] Using OpenGL 3.0
[29/05/2024 01:19:40.000] [INFO] Loading icons
[29/05/2024 01:19:40.000] [INFO] Loading band plans
[29/05/2024 01:19:40.000] [INFO] Loading band plans color table
[29/05/2024 01:19:40.000] [INFO] Loading modules
[29/05/2024 01:19:40.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/airspyhf_source.dll
[29/05/2024 01:19:40.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/airspy_source.dll
[29/05/2024 01:19:40.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/audio_sink.dll
[29/05/2024 01:19:40.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/audio_source.dll
[29/05/2024 01:19:40.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/bladerf_source.dll
[29/05/2024 01:19:40.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/discord_integration.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/file_source.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/frequency_manager.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/hackrf_source.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/hermes_source.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/iq_exporter.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/limesdr_source.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/m17_decoder.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/meteor_demodulator.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/network_sink.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/perseus_source.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/plutosdr_source.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/radio.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/recorder.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/rfspace_source.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/rigctl_client.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/rigctl_server.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/rtl_sdr_source.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/rtl_tcp_source.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/scanner.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/sdrplay_source.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/sdrpp_server_source.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/spyserver_source.dll
[29/05/2024 01:19:41.000] [INFO] Loading C:/Program Files/SDR/sdrpp_windows_x64/modules/usrp_source.dll
[29/05/2024 01:19:41.000] [INFO] Initializing Airspy Source (airspy_source)
[29/05/2024 01:19:41.000] [INFO] Initializing AirspyHF+ Source (airspyhf_source)
[29/05/2024 01:19:41.000] [INFO] Initializing Audio Sink (audio_sink)
[29/05/2024 01:19:41.000] [INFO] Initializing Audio Source (audio_source)
[29/05/2024 01:19:41.000] [INFO] Initializing File Source (file_source)
[29/05/2024 01:19:41.000] [INFO] Initializing Frequency Manager (frequency_manager)
[29/05/2024 01:19:41.000] [INFO] Initializing HackRF Source (hackrf_source)
[29/05/2024 01:19:41.000] [INFO] Initializing Network Sink (network_sink)
[29/05/2024 01:19:41.000] [INFO] Initializing PlutoSDR Source (plutosdr_source)
[29/05/2024 01:19:41.000] [INFO] Initializing RFspace Source (rfspace_source)
[29/05/2024 01:19:41.000] [INFO] Initializing RTL-SDR Source (rtl_sdr_source)
[29/05/2024 01:19:41.000] [INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[29/05/2024 01:19:41.000] [INFO] Initializing Radio (radio)
[Resamp] predec: 32, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[Resamp] predec: 32, interp: 5000, decim: 7813, inacc: 0.000000%, taps: 593750
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[29/05/2024 01:19:41.000] [WARN] Demod switch took 17557 us
[29/05/2024 01:19:41.000] [INFO] RtAudio stream open
[29/05/2024 01:19:41.000] [INFO] Initializing Recorder (recorder)
[29/05/2024 01:19:41.000] [INFO] Initializing Rigctl Server (rigctl_server)
[... cut ...]
[29/05/2024 01:20:53.000] [INFO] New DSP samplerate: 1228800.000000 (source samplerate is 1228800.000000)
[29/05/2024 01:20:53.000] [WARN] End
[29/05/2024 01:20:54.000] [INFO] RFSpaceSourceModule 'RFspace Source': Start!
[29/05/2024 01:20:54.000] [INFO] RFSpaceSourceModule 'RFspace Source': Tune: 2200000.000000!
[29/05/2024 01:20:59.000] [INFO] RFSpaceSourceModule 'RFspace Source': Stop!

C:\Program Files\SDR\sdrpp_windows_x64>

https://github.com/AlexandreRouma/SDRPlusPlus/assets/5446930/eb242d64-413c-4b64-9333-efba916508a7

AlexandreRouma, 2024-05-28:
Try closing every other application and try to connect to the CloudSDR while recording the network traffic with Wireshark (google how to do this if you're not familiar with that tool).
Make sure to stop the SDR quickly if does start streaming data because that can make the recording huge and email me the recording at `alexandre.rouma@gmail.com` (don't post it publically, it might contain sensitive info).

AlexandreRouma, 2024-05-28:
Thanks for the capture, I can see exactly what's going wrong but I'm not sure what's causing it at this stage.
Seems that the UDP receive code is not getting the packets even though they are clearly being received by your computer.