AlexandreRouma/SDRPlusPlus #1415: Error 0xc0000142 on windows 11

EA4GSX, 2024-06-07:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel(R) Core(TM) i5-6500T CPU @ 2.50GHz   2.50 GHz
- RAM: 16,0 GB (15,9 GB usable)
- GPU: Intel(R) HD Graphics 530

- SDR: RTL-SDR V3 USB LOCAL

**Software**
- Operating System: Windows 11 Pro+ 22000.2538 (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: 1.2.0 Version +06/06/2024 Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**
Wen open file sdrpp.exe apear error 0xc0000142 
**Steps To Reproduce**
1. Doble click in file sdrpp.exe...
2. Error apear 0xc0000142 
3. Acept and close windows error

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
[image]).

**Additional info**
- System 

  - Provider 

   [ Name]  Application Error 
 
  - EventID 1000 

   [ Qualifiers]  0 
 
   Version 0 
 
   Level 2 
 
   Task 100 
 
   Opcode 0 
 
   Keywords 0x80000000000000 
 
  - TimeCreated 

   [ SystemTime]  2024-06-07T14:19:10.4464858Z 
 
   EventRecordID 74872 
 
   Correlation 
 
  - Execution 

   [ ProcessID]  7604 
   [ ThreadID]  0 
 
   Channel Application 
 
   Computer SDR 
 
   Security 
 

- EventData 

   sdrpp.exe 
   0.0.0.0 
   6661f650 
   MSVCP140.dll 
   14.34.31938.0 
   e7a4b09e 
   c0000005 
   0000000000013028 
   1d88 
   01dab8e5a9defcb9 
   C:\Users\echoa\Desktop\sdr++\sdrpp_windows_x64\sdrpp.exe 
   C:\WINDOWS\SYSTEM32\MSVCP140.dll 
   c862c04c-0cbe-4a60-94dc-9ba9856b8973

bebepip, 2024-06-07:
Hi , same problem with windows10 22H2, not working last SDR++ Nightly Build, pls can u check it ?
[image]

AlexandreRouma, 2024-06-26:
Should be fixed, let me know if not.