AlexandreRouma/SDRPlusPlus #1417: June 7 nightly build crashes

delboy85341, 2024-06-08:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel i5-3570 CPU @ 3.40GHz
- RAM: 32GB
- GPU: Nvidia GT-710
- SDR: (Remote or local? If remote, what protocol?) RSP1A

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15) Kubuntu 22.04.4 LTS
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).
Version: 1.2.0 nightly build June 7 2024
**Bug Description**
A clear description of the bug.

When I tried to start it, the app started as it normally does until it got to the part that said Loading band plans.  Then it crashed.  I launched it from a terminal and got the following:

$ sdrpp
[07/06/2024 20:00:50.000] [INFO] SDR++ v1.2.0
[07/06/2024 20:00:50.000] [INFO] Loading config
[07/06/2024 20:00:50.000] [INFO] Using OpenGL 3.0
[07/06/2024 20:00:50.000] [INFO] Loading icons
[07/06/2024 20:00:50.000] [INFO] Loading band plans
terminate called after throwing an instance of 'nlohmann::detail::parse_error'
  what():  [json.exception.parse_error.101] parse error at line 429, column 9: syntax error while parsing array - unexpected '{'; expected ']'
Aborted (core dumped)

**Steps To Reproduce**
1. Start SDR++  No other steps are needed.
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

I tried  `sdrpp -c` with the following result:
bash: [07/06/2024: No such file or directory

Next I entered sdrpp -c and got the following:
sdrpp -c
[07/06/2024 20:15:28.000] [INFO] SDR++ v1.2.0
Unknown argument
-a --addr               Server mode address
   --autostart          Automatically start the SDR after loading
-h --help               Show help
-p --port               Server mode port
-r --root               Root directory, where all config files are stored
-s --server             Run in server mode

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

delboy85341, 2024-06-08:
I   was using version 3.14.0 of the api when I first got the error.  I upgraded to 3.15.2 and got the same error.

AlexandreRouma, 2024-06-08:
Fixed, was a bad pull request from someone who didn't check check the syntax of what they were submitting.
let me know if it's not fixed.

closed 2024-06-08 by AlexandreRouma