AlexandreRouma/SDRPlusPlus #1422: Limited Sampling Rate

Paloma-96, 2024-06-12:
**Hardware**
- SDR: USRP B210 connected locally 

**Software**
- Operating System: Ubuntu 22.04
- SDR++: v1.1.0

**Bug Description**
Even though the maximum supported sampling rate is higher, the maximum Sample Rate in GUI is 16 MHz, both using USRP and SoapySDR driver.

How can I increase the max sampling rate?

zekiispa, 2024-06-12:
Hello ,
Normally the device promises 56MHZ bandwidth. I have the same problem with the USRP B200mini. I cannot see more than 16mhz bandwidth on the screen.

AlexandreRouma, 2024-06-13:
Not a bug, base clock selection just hasn't been implemented yet. I'll change this to a feature request.

zekiispa, 2024-06-13:
Got it, thanks.

alex0com, 2024-08-17:
I've noticed the same thing with a USRP B210mini clone. Running some test code to query the device, UHD reports
usrp->get_rx_bandwidth_range is 200000.00 Hz to 56000000.00 Hz
usrp->get_rx_rates is 31250.00 S/s to 16000000.00 S/s

SDR++ looks to be using the correct values provided from the driver. I'm guessing this may be due to the custom FPGA and fw provider by the clone manufacturer? 

If you set the master clock rate manually instead of automatic, the 56MHz a a few others show up. Or if I change the code to populate the srList with the get_rx_bandwidth_range instead, I get SDR++ to show the full 2Hz-56MHz sample rate. This seems to work. But I have no idea what I'm doing, I'm just playing around with this. 

https://github.com/AlexandreRouma/SDRPlusPlus/blob/118e1fbff00df8b91b809f05be386cfcc3fd4831/source_modules/usrp_source/src/main.cpp#L142
Change the above line to this if you want the full sample rate list on the clone
`auto srList = dev->get_rx_bandwidth_range(chanId);`

AlexandreRouma, 2024-08-17:
Bandwidth is for the input filter, has nothing to do with available samplerates. As I said in my last response, the reason the samplerate is limited to that value is the baseclock is set to 16MHz. It has to be dynamically adjusted but I have yet to determine an algorithm to select the best options.

alex0com, 2024-08-17:
> Bandwidth is for the input filter, has nothing to do with available samplerates. As I said in my last response, the reason the samplerate is limited to that value is the baseclock is set to 16MHz. It has to be dynamically adjusted but I have yet to determine an algorithm to select the best options.

I apologize for my ignorance, I'm just playing around with all of this and only wanted to see the full bandwidth in SDR++ for while scanning. I realize I'm showing a round peg into a squre hole with what I did.

I found this mentioned on the Ettus website: "The default master clock rate defaults to 32 MHz" for a B200/210. As I'm using the B200 knockoff, I just assumed it was the custom  fw and FPGA. The genuine Ettus images do not work on this device. I was guessing genuine images would report correctly. Or at least set the "automatic" to 32 instead of 16.
https://files.ettus.com/manual_archive/release_003_008_005/html/page_usrp_b200.html

When SDR++ starts, it shows it setting master clock rate as "automatic" and setting to 16.0MHz. A short snippet querying the device using only uhd/usrp/multi_usrp.hpp  and get_rx_rates returned the max sample rate of 16MHz. But setting it manually worked, so I just shoved the range into the dropdown.  

`[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 56.000000 MHz... 
[16/08/2024 20:51:54.000] [WARN] ConfigManager locked, waiting...
[INFO] [B200] Actually got clock rate 56.000000 MHz.
[Resamp] predec: 1024, interp: 1500, decim: 1709, inacc: 0.000000%, taps: 129882
[16/08/2024 20:51:55.000] [INFO] New DSP samplerate: 56000000.000000 (source samplerate is 56000000.000000)` 

I realize this is wrong but it does allow me to watch the full 56MHz waterfall. :) Thank you for your awesome work!

alex0com, 2024-08-17:
My apologies, previous posts were wrong and I deleted them.

Setting the master clock rate manually allows me to view the full bandwidth using the B210mini clone instead of just 16MHz. What effect this may have beyond just watching the waterfall is beyond my level of understanding. 

```
    // Set the master clock rate to 56 MHz (or adjust as necessary)
    double master_clock_rate = 56e6; // 56 MHz
    dev->set_master_clock_rate(master_clock_rate);
    double actual_clock_rate = dev->get_master_clock_rate();
    if (actual_clock_rate != master_clock_rate) {
        flog::warn("Requested clock rate of 56 MHz, but actual clock rate is {0} MHz", actual_clock_rate / 1e6);
    }
    
```
    
```
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[16/08/2024 21:50:53.000] [WARN] ConfigManager locked, waiting...
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 56.000000 MHz... 
[INFO] [B200] Actually got clock rate 56.000000 MHz.
```

Thanks again for your great work!

zekiispa, 2024-10-25:
I want to modify this code you shared below to manually set the master clock rate to 28mhz. But where exactly do I paste it?

    // Set the master clock rate to 56 MHz (or adjust as necessary)
    double master_clock_rate = 56e6; // 56 MHz
    dev->set_master_clock_rate(master_clock_rate);
    double actual_clock_rate = dev->get_master_clock_rate();
    if (actual_clock_rate != master_clock_rate) {
        flog::warn("Requested clock rate of 56 MHz, but actual clock rate is {0} MHz", actual_clock_rate / 1e6);
    }

alex0com, 2024-10-25:
> I want to modify this code you shared below to manually set the master clock rate to 28mhz. But where exactly do I paste it?
> 

I added it to source_modules/usrp_source/src/main.cpp on line 139. Again, this is only to allow you to select the higher value, no idea about best use case and performance. 
here:
https://github.com/AlexandreRouma/SDRPlusPlus/blob/3982db73d3f9d27972feaff819fe08e9ec7b4c82/source_modules/usrp_source/src/main.cpp#L139

zekiispa, 2024-10-26:
Hello Alex,

I have my usrp b205 device. Thanks for the code. I experimented with 28mhz, 40mhz, 56mhz. At these bandwidths, the waterfall visually flows smoothly. But when I select a channel in the FM radio band and listen, the sound is intermittent. I guess it doesn't work properly in this way :( did you do a test like this? For example, did you try listening to FM radio with 28mhz or 56 baseclock? 

I think AlexandreRouma will solve this when the time comes :) we are waiting patiently.

alex0com, 2024-10-26:
> Hello Alex,
>
> I have my usrp b205 device. Thanks for the code. I experimented with 28mhz, 40mhz, 56mhz. At these bandwidths, the waterfall visually flows smoothly. But when I select a channel in the [...]

Mine works great and I have no issues with sound unless I select 56 and get overruns at high bandwidth. I’ve added extra code to populate the full range of samplerate and bandwidth. The only weird thing I see is some aliasing on certain selections.

zekiispa, 2024-10-26:
Hello again Alex, which driver version are you using, can you share the link if possible (files.ettus.com/binaries/uhd/)

Also, if possible, can you share main.cpp with me?

zeki_ispa@hotmail.com

alex0com, 2024-10-26:
I'm on MacOS 15 arm64 using the UHD that's available via brew.
==> uhd: stable 4.7.0.0 (bottled), HEAD
My b210 fpga bin is a custom build by the Chinese seller of the clone. I never got around to finding the differences in the fpga code between the one provided by the seller and the binaries provided directly from USRP. The latter do not work on this device.

Here are my changes. Use at your own risk
https://pastebin.com/AMyXs9Ab