AlexandreRouma/SDRPlusPlus #1424: Support for new SDRplay RSP models: RSP1B and RSPdx-R2

fventuri, 2024-06-15:
I made a few changes to `source_modules/sdrplay_source/src/main.cpp` to support the new SDRplay RSP models: RSP1B and RSPdx-R2
The source file with my changes is attached, if you would like to use it for the official version of SDR++.

I also have a branch called `rsp1b+rspdx-r2` with these changes: https://github.com/fventuri/SDRPlusPlus/tree/rsp1b%2Brspdx-r2 - if you prefer, I can create a pull request.

Regards,
Franco Venturi

[sdrplay_source.tar.gz](https://github.com/user-attachments/files/15849652/sdrplay_source.tar.gz)

AlexandreRouma, 2024-06-15:
Duplicate of #1311.

As I already said in that and other threads, I'm not adding a bunch of device specific code to support these without having a unit to test myself. I'm the one who will receive the angry emails when something doesn't work, not SDRplay. I'll add support myself when I have the budget for these SDRs, so don't expect it any time soon.

closed 2024-06-15 by AlexandreRouma

fventuri, 2024-06-15:
Alex, I fully understand your concerns and your timeline, since you must be very busy with this excellent program.

I'll try to keep my fork up to date with the official version for those who would like to run SDR++ with their RSP1B and RSPdx-R2.

Thanks for all your hard work with SDR++!

Franco

starvald, 2024-10-11:
@fventuri i'm using your patch and its working fine with my rsp1b btw.

jhnptr, 2025-02-06:
I feel there's a missed opportunity to leverage @fventuri 's code. Since the device support is a module, couldn't it be added in a separate "experimental-not-supported" module that's included in the daily builds?

I realize that you want to maintain control over the software and maintain a certain level of quality, but there surely must be a way to find a middle ground where the community can help you in supporting devices you don't have.

I'd really like to use SDR++ with my RSPdx-R2.

fventuri, 2025-02-06:
My understanding from this post by Alexandre https://groups.io/g/SDRPlayUsers/message/12752 is that SDR++ now supports the new RSP models natively, so there should be no reason to use my code now.

Franco

jhnptr, 2025-02-06:
> My understanding from this post by Alexandre https://groups.io/g/SDRPlayUsers/message/12752 is that SDR++ now supports the new RSP models natively, so there should be no reason to use my code now.
> 
> Franco

Thank you @fventuri ! I completely missed it. I hadn't installed the SDRPlay API so the source wasn't listed in the sources even if the module was loaded.

And @AlexandreRouma thank you for adding the newer model support and keep up the great work!