AlexandreRouma/SDRPlusPlus #1427: MAC OSX SONOMA 14.5 fix DEVICE problem

SandorPeteer, 2024-06-23:
My RTL-SDR v4 device is not working with Sonoma 14.5

I've been using my RTL-SDR v4 device with SDR++ for a while now and it's been working great. However, after updating to the latest macOS Sonoma 14.5, SDR++ is no longer able to detect or use the device. I've tried everything I can think of, but nothing seems to work.

I've tried the following:

Restarting my computer
Reinstalling SDR++
Checking the USB connection
Trying a different USB port
I've also checked the SDR++ forums and GitHub page, but I haven't been able to find any solutions.

I'm at my wit's end here. I'm hoping that someone in the community might have a solution for me. I would really appreciate any help that you can provide.

Thank you for your time.

SP

SandorPeteer, 2024-06-23:
Sorry!

****I found the solution!****

I was able to get my RTL-SDR v4 device working with SDR++ on macOS Sonoma 14.5 by following these steps:

1. Open a Terminal window.
2. Run the following command:

```
brew install libusb fftw glfw airspy airspyhf portaudio hackrf rtl-sdr libbladerf codec2 zstd
pip3 install mako
```

This will install all of the necessary dependencies for SDR++.

3. Once the installation is complete, restart your computer.
4. Launch SDR++.
5. The RTL-SDR v4 device should now be listed in the Devices menu.
6. Select the device and click the "PLAY" button.

SDR++ should now be able to detect and use the RTL-SDR v4 device.

**I'm so happy that I was able to find a solution to this problem!**

I hope this helps other people who are having the same issue.

**SP**

AlexandreRouma, 2024-06-23:
No you're not supposed to have to do that...
I don,'t know what's wrong, I don't have a modern mac, but this is definitely not the solution.
What build of SDR++ are you running?

SandorPeteer, 2024-06-24:
> No you're not supposed to have to do that... I don,'t know what's wrong, I don't have a modern mac, but this is definitely not the solution. What build of SDR++ are you running?

SDR++ 1.2.0.
 
Maybe the USB-C cable has a contact fault. 

But now, everything ok!

closed 2024-06-24 by SandorPeteer