AlexandreRouma/SDRPlusPlus #1428: Crash sig 11 bad access

rbur004, 2024-06-26:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU:  Apple M1 Pro
- RAM: 16G
- GPU:  M1 16 Core
- SDR: RTL-SDR clone (Radarbox yellow VHF stick)

**Software**
- Operating System: MacOS Sonoma 14.5
- SDR++: 1.2.0 20:06:18 June 4 2024

**Bug Description**
Crashing sometimes on startup, sometimes while I'm using it.
Mostly trying to access 0, but sometimes other addresses
e.g. KERN_INVALID_ADDRESS at 0x2220202020202020 
my guess is a pointer isn't initialised before it is being used

**Steps To Reproduce**
1. Normal use. Doesn't seem to have a specific trigger

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.
[crash-dumps.zip](https://github.com/user-attachments/files/15981580/crash-dumps.zip)

AlexandreRouma, 2024-10-07:
The first type of crash is a libusb bug so nothing I can do on my side.
The second is a bug in the portaudio sink module that attempts to close the stream even if it failed to open. But clearly it failing to open shows there is another issue (so again, I need for information).
The third is a random crash in the GUI, and again, I have no information on what you did so I can't do anything.

AlexandreRouma, 2025-02-21:
No further information was provided so i'll close this issue as stalled.

closed 2025-02-21 by AlexandreRouma