AlexandreRouma/SDRPlusPlus #1431: SDRPlay not working on MacOS ARM.

jaminmc, 2024-07-02:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Apple M1 Ultra
- RAM: 64 GB
- GPU: Apple M1 Ultra
- SDR: RSP1

**Software**
- Operating System: MacOS 14.5 (23F79)
- SDR++: v1.2.0 (Built at 15.43.50, Jun 26 2024
**Bug Description**
SDR Play is unavailable.  

**Steps To Reproduce**
1. If I start from the commandline, I see that it doesn't read the driver.
```
[01/07/2024 18:37:39.000] [INFO] Loading /Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib
[01/07/2024 18:37:39.000] [ERROR] Couldn't load /Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib: dlopen(/Applications/SDR++.app/Contents/Resources/../Plugins/sdrplay_source.dylib, 0x0005): Library not loaded: @rpath/libsdrplay_api.so.3
  Referenced from: <60A2F4F0-2259-38D7-ABAC-ED05C91FDEC9> /Applications/SDR++.app/Contents/Plugins/sdrplay_source.dylib
  Reason: tried: '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/Plugins/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/Frameworks/../Frameworks/libsdrplay_api.so.3' (no such file), '/Applications/SDR++.app/Contents/MacOS/../Frameworks/libsdrplay_api.so.3' (no such file)

```

## Workaround:
I got it to work if I:
```
ln -s /usr/local/lib/libsdrplay_api.so.3 /Applications/SDR++.app/Contents/Frameworks 
```

tonycart, 2024-07-18:
Thank you for this information. I've had the same problem.

drewzh, 2024-10-15:
This symlink fixed it for me also. Issue reproduced on 3 different setups.