AlexandreRouma/SDRPlusPlus #1434: rigctl_server unable to evaluate mode change bandwidth string

guitarpicva, 2024-07-06:
**Bug Description**
misc_modules/rigctl_server/src/main.cpp

~ line 450 the Mode setting string for the bandwidth is evaluated for being all digits or
the first char being '-'.  Looping through the std::string for chars seems to also eval 
the closing NULL of the c_str and so the evaluation always fails.

**Steps To Reproduce**
1. start sdrpp
2. telnet to running rigctl_server on the sdrpp
3. send mode change command e.g. M AM 6000

Eliminating that section of the code allows properly formatted commands to work.

AlexandreRouma, 2024-07-06:
Works fine for me. You're probably trying to connect with telnet which adds extra terminal control junk.
[image]
[image]

closed 2024-07-06 by AlexandreRouma

guitarpicva, 2024-07-06:
You are perhaps correct, though I did try raw TCP socket from PuTTY as well.

guitarpicva, 2024-07-07:
Tested your way and worked correctly with original code.   I guess I"m used to using Hamlib  rigctl via a telnet session.  Sorry to bother!!

guitarpicva, 2024-07-09:
adding the line marked "// MW remove...." seems to tolerate telnet sessions.  It 
seems it was only the carriage returns causing angst with normal linux telnet.  Not 
really important, but I was curious. :)

// Split command into parts and remove excess spaces
        for (char c : cmd) {
            if(c == '\r') {continue;} // MW remove carriage returns from things like telnet
            if (lastWasSpace && c == ' ') { continue; }
            else if (c == ' ') {
                parts.push_back(corr);
                corr.clear();
                lastWasSpace = true;
            }
            else {
                lastWasSpace = false;
                corr += c;
            }
        }