AlexandreRouma/SDRPlusPlus #1444: Scanner: How to scan specific frequencies?

agentpatience, 2024-08-09:
Can someone tell me how to use SDR++ to allow you to manually enter frequencies and scan just those frequencies?

Axpelle, 2024-08-11:
It appears that for the time being you can only scan a range of frequencies between two preset frequencies. It's not yet possible to scan a bank of individual frequencies.

AlexandreRouma, 2024-08-19:
planned but no ETA

Sobollion, 2024-09-08:
Really need functions like freqmgrsuite.com in sdrpp

Axpelle, 2024-09-09:
When SDR++ eventually makes granular scanning available it will most likely be a home-grown solution rather than a transplant of existing plugins.

> Really need functions like freqmgrsuite.com in sdrpp

jthartika, 2024-09-20:
I would be interested also. In particular in addition to scanning certain frequencies I would like the option for inverse, i.e. ignore certain frequencies.

osuismalife, 2025-02-20:
was this implemented yet?