AlexandreRouma/SDRPlusPlus #1446: how to upgrade SDR++ to v1.2.0 on intel MacBook

RickReale, 2024-08-09:
I have SDR++ v1.1.0 and wish to upgrade to 1.2.0.
2017 MacBook 12' Retina running Monterey 12.7.5.
Download won't install. Get this:

[image]

Don't remember just how I installed my original version last year or so, but do recall there was some other steps involving some other app to make it all happen.
Anyone here able to help me out with this?
Cheers.

RickReale, 2024-08-10:
Further to above. It seems I have MacPorts installed but app still won't install.
Assuming I get there, will the Bookmarks and other settings be retained after the upgrade to 1.2.0?

Axpelle, 2024-08-11:
This is a precaution against malware. SDR++ is legit but you must grant it permission to be opened. Google how to do this.

RickReale, 2024-08-11:
Thanks Axpelle. Got it sorted and 1.2.0 is now installed and running well.

closed 2024-08-14 by AlexandreRouma