AlexandreRouma/SDRPlusPlus #1453: SDR++-server no connect on X64 but ARM okay

loewal, 2024-08-16:
**Hardware**
- CPU: ARM M2
- RAM: 8GB
- GPU: ...
- SDR: SDR++-server

**Software**
- Operating System: Macos Sequoia beta 2
- SDR++: 1.2.0 built on Aug 5
**Bug Description**

No connect to SDR++-server
**Steps To Reproduce**
1. select ip 10.0.0.19
2. select port 5259
3. connect: fail on X64, okay on ARM version. Worked on older versions

Starting from terminal with /path/to/the/SDR++.app/Contents/MacOS/sdrpp works!
Removes link on dock and placed again, but no success.
Still from prompt working.

* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**
No error...nothing happens

KubaPro010, 2024-08-16:
Maybe the x86 machine has no connection or different connection? (make sure it's the same wifi), also what's the error?

loewal, 2024-08-16:
Cannot connect to my Raspiberry Pi SDR++ server from within the app I called via Apllications.
When using the same app, but through packagecontents, macos, etc. and then ./sdrpp the connection can be made.
Earlier versions were okay.

loewal, 2024-08-16:
Tested on an older Macbook Pro (Intel) and there it works. Maybe a Seqouia (beta 2) related problem?

loewal, 2024-08-16:
Deleted the app, reinstalled and created an alias.
Working from the alias works! The direct way does not.
Perhaps a rights related problem?

AlexandreRouma, 2024-08-19:
Seems like some MacOS security issue. You probably need to grant it the permission to run normally since it was downloaded off the internet. I don't have any Arm mac or any x86 mac that can run such a recent version of MacOS

loewal, 2024-08-20:
Downloaded latest versions of X64 and ARM.
Started Arm-version and connected to SDR++-server. OK, then stopped and left app.
Started X64-version and tried to connect: server already in use......
Have you changed this behavior?

loewal, 2024-08-20:
Addendum:
After leaving the X64-version started the ARM-version again and now no connect anymore.
So, the server-connection is not closed completely/the right way?

closed 2026-10-22 by AlexandreRouma