AlexandreRouma/SDRPlusPlus #1454: Spectrum Analyzer and Waterfall Freeze After a Couple Seconds

psoltan, 2024-08-19:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU:  Intel i5-4570 (Dell Optiplex 9020)
- RAM: 16GB
- GPU:  none
- SDR: local: LimeSDR-USB (x2 Rx and x2 Tx)
- 
**Software**
- Operating System: Linux DragonOS FocalX R35
- SDR++: v1.2.0 (Built at 18:01:54, 18 August 2024)

**Bug Description**
The spectrum display and waterfall freeze after a couple seconds.  The program doesn't crash and I can continue stopping and starting the program but the spectrum and waterfall don't work.

I ran the program from the build directory with "./sdrpp -r ../root"

**Steps To Reproduce**
1. ...
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
[image]

**Additional info**
Add any other relevant information.

AlexandreRouma, 2024-08-19:
No steps to reproduce, no logs... What am I supposed to do here?

psoltan, 2024-08-19:
Hi Alexandre,

Did you even try to build the latest 1.2.0 version?  I ran it from an xterm
window and there are no errors or warnings when the spectrum/waterfall
freeze.  How is that possible?

Phillip