AlexandreRouma/SDRPlusPlus #1455: SoapySDR source not included in make_mac_bundle.sh so not included in release build

vk1kcm, 2024-08-23:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Apple M1 Max
- RAM: 32GB
- GPU: Apple M1 Max
- SDR: Local

**Software**
- Operating System: macOS 14.5.1
- SDR++:  1.2.0 Git Pull at 0400 UTC, 23/8/2024.

**Bug Description**
The make_macos_bundle.sh script doesn't copy in the soapysdr source module, so SoapySDR is not available in the build, nor in the released binary on GitHub.

**Steps To Reproduce**
1. Build sdrpp
2. Try to use SoapySDR
3. Fail

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
To fix I just added the following to the make_macos_bundle.sh script:
```
bundle_install_binary $BUNDLE $BUNDLE/Contents/Plugins $BUILD_DIR/source_modules/soapy_source/soapy_source.dylib
```

AlexandreRouma, 2024-08-23:
The SoapySDR source module has been depracted for months.
I don't know what hardware you have but you most likely should NOT use it.

closed 2024-08-23 by AlexandreRouma

daochild, 2024-09-08:
> The SoapySDR source module has been deprecated for months. I don't know what hardware you have but you most likely should NOT use it.

Could you please tell me what the last commit supports this module?

helioshk, 2025-05-07:
Has somebody found out which last commit was supporting soapy?