AlexandreRouma/SDRPlusPlus #1456: V1.20 Ubuntu Noble Release Doesn't find RTL-SDR v4 as a source

jnorrid, 2024-08-26:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel(R) Core(TM) i7-7700K CPU @ 4.20GHz
- RAM: 32GB
- GPU: MSI NVIDIA 1080
- SDR: Local RTL-SDR V4 - Drivers installed correctly and rtl_test functions fine.

**Software**
- Operating System: Ubuntu 24.04 (Noble) Fully updated as of today (08/25/2024)
- SDR++: 1.20 for Ubuntu Noble (24.04) pulled from repo releases page on 08/25/2024

**Bug Description**
I followed all instructions to install the RTL-SDR V4 drivers including blacklisting prior versions of the driver. I can confirm that the RTL-SDR V4 dongle is properly functioning with rtl_test as well as with GQRX on this machine. When I install SDR++ from the releases page for 1.20 using the Ubuntu Noble .deb file, the application comes up fine, however, the RTL-SDR does NOT show up as an available source - it is simply not in the pick list.

**Steps To Reproduce**
1. Install RTL-SDR V4 drivers on a clean machine including proper blacklisting of prior drivers.
2. Reboot the system so that all changes take effect.
3. Install and run SDR++ from the 1.20 releases page for Ubuntu Noble after connecting the dongle and running rtl_test.

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
I am wondering if the RTL-SDR V4 was somehow left out of the soapysdr plugins for this build, just a wild guess. Again, GQRX works fine and I am able to tune, get waterfalls, enable hardware AGC, etc. so I know that this is not a Linux driver issue.

AlexandreRouma, 2024-08-26:
show the logs

AlexandreRouma, 2024-08-26:
And also exactly *which* drivers did you install, link to them.
Without checking I bet it's a varient of the osmocomm ones that have a different major version which won't link to SDR++ because it was compiled for the version that ships with the OS.

jnorrid, 2024-08-26:
I built the drivers per the "Getting Started With Linux" section of the RTL-SDR V4 online documentation here: https://www.rtl-sdr.com/rtl-sdr-quick-start-guide/

The relevant "Getting Started With Linux" section explains the process. Based on the GH link in this procedure, the source for the drivers gets cloned locally from: https://github.com/rtlsdrblog/rtl-sdr-blog

After removing (dpkg --purge <name of deb file) and re-installing, now everything works and RTL-SDR shows up. I am going to try this on a couple other machines in my shop and will come back if any issues. Only thing I can think is that Noble is a TERRIBLE release and I had several packages that were stuck in phasing for update. Between attempts today, I performed a sudo apt update && sudo apt upgrade for all of my repos, so perhaps there was a package in there that was malfunctioning. NOBLE has been the most unreliable LTS release I have seen in 12 years. Thank you for your patience.

AlexandreRouma, 2024-08-26:
I'll assume this was a messed up install, probably because the drivers were installed on top of the existing ones. I haven't gotten any other reports of issues on Noble in the few months that the deb has been available.

closed 2024-08-26 by AlexandreRouma