AlexandreRouma/SDRPlusPlus #1458: GCC14 compile error with RYFI Decoder: error: ‘NULL’ was not declared in this scope

jharvell, 2024-08-26:
**Hardware**
- CPU: **N/A**
- RAM: **N/A**
- GPU: **N/A** 
- SDR: (Remote or local? If remote, what protocol?) **N/A**

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15) **Gentoo Base System release 2.15**
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

_commit fe9ac6c9a1cbb1808c2e4aa53ea3061c74d3c656_

**Bug Description**
Compile error in decoder_modules/ryfi_decoder/src/ryfi/packet.cpp

**Steps To Reproduce**
1. git checkout fe9ac6c9a1cbb1808c2e4aa53ea3061c74d3c656
2. mkdir build
3. cd build
4. cmake -DOPT_BUILD_AIRSPYHF_SOURCE=OFF -DOPT_BUILD_RYFI_DECODER=ON ..
5. make -j

Observe the following compile error:
```
In file included from /mnt/etretat/home/jharvell/akita/git/SDRPlusPlus/decoder_modules/ryfi_decoder/src/ryfi/packet.cpp:1:
/mnt/etretat/home/jharvell/akita/git/SDRPlusPlus/decoder_modules/ryfi_decoder/src/ryfi/packet.h:85:29: error: ‘NULL’ was not declared in this scope
   85 |         uint8_t* _content = NULL;
```

I fixed with the attached patch using the preferred C++ keyword `nullptr` that is available since C++11.  If you want to support a language standard prior to C++11, then you could alternatively add `#include <cstddef>` to get the definition of `NULL`.  I also changed the `#include <stdtint.h>` include to the C++ `#include <stdint>` mechanism.

[sdrpp-ryfi-nullptr.patch.txt](https://github.com/user-attachments/files/16753147/sdrpp-ryfi-nullptr.patch.txt)

AlexandreRouma, 2024-08-26:
`nullptr` and `cstdint` are stupid and ugly C++ features which I refuse to use.
I'll just `#include <stddef.h>`

AlexandreRouma, 2024-08-26:
done

closed 2024-08-26 by AlexandreRouma

jharvell, 2024-08-26:
Wow, that was fast....thanks!  FYI I created a Gentoo package for this.  I'll let you know if/when it gets accepted by the Gentoo devs.

AlexandreRouma, 2024-08-26:
If you package it make sure no code is changed and all the modules listed as "Built in releases" in  the [module list](https://github.com/AlexandreRouma/SDRPlusPlus#module-list) are included in the package.