AlexandreRouma/SDRPlusPlus #1460: Unable to build and use both AUDIO and PORTAUDIO sinks

jharvell, 2024-08-28:
**Hardware**
- CPU: N/A
- RAM: N/A
- GPU: N/A
- SDR: (Remote or local? If remote, what protocol?) N/A

**Software**
- Operating System: Gentoo Base System release 2.15
- SDR++: 6dce28345c4b72576fdd32496a00781e73794b57

**Bug Description**
I am working on getting this packaged for Gentoo.  As you requested, I am making sure all the "Built In Release" modules are included.  However, it is not possible to build with both OPT_BUILD_AUDIO_SINK and OPT_BUILD_PORTAUDIO_SINK enabled because they both use the same CMAKE project name _audio_sink_.

Looking at the code, it seems like only one or the other is intended to be built, which would imply the module documentation is wrong.  But assuming the documentation is correct and the code is wrong, I have attached a patch that enables both to be built.  This patch also registers both both modules so that they are included in the drop down list Sinks->Radio->Audio.

I also noticed that the AUDIO module does not implement configuration.  So the audio device last selected with that sink is not persisted like it is with AUDIO and NEW_PORTAUDIO.  Since those are still beta, you might not want to take that part.  But I include it here as demonstration that everything works at runtime with these changes.

**Steps To Reproduce**
1. cd to top level directory
2. ```mkdir build```
3. ```cd build```
4. ```cmake -DOPT_BUILD_AIRSPYHF_SOURCE=OFF -DOPT_BUILD_AUDIO_SINK=ON -DOPT_BUILD_PORTAUDIO_SINK=ON ..```

Observe the following error from CMake v3.30.2:
```
CMake Error at sdrpp_module.cmake:10 (add_library):
  add_library cannot create target "audio_sink" because another target with
  the same name already exists.  The existing target is a shared library
  created in source directory
  "/mnt/etretat/home/jharvell/git/SDRPlusPlus/sink_modules/audio_sink".  See
  documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  sink_modules/portaudio_sink/CMakeLists.txt:6 (include)
```

Note I only turn off AIRSPYHF since I don't yet have the library it depends on.
[upstream_audio_portaudio.patch.txt](https://github.com/user-attachments/files/16790068/upstream_audio_portaudio.patch.txt)

AlexandreRouma, 2024-08-29:
On purpose, you're not supposed to use both.

closed 2024-08-29 by AlexandreRouma

jharvell, 2024-08-29:
Ok, that's what I thought.  So I cannot include both in the release as you suggested.  Which (if either) should I include?

AlexandreRouma, 2024-08-29:
The module list in the readme is a bit ambiguous on this subject because the MacOS release uses the portaudio based `audio_sink` module, the Android release uses the AAudio based one and the windows and linux releases use the RtAudio one. In addition to this, there is the "New portaudio sink" which is supposed to be the replacement for the RtAudio one but is currently in testing only.
This means you must include the RtAudio based `audio_sink` and the `new_portaudio_sink` modules.

jharvell, 2024-08-29:
Thanks for the clarification.

jharvell, 2024-08-29:
I'v submitted a pull request to add the package to Gentoo.  Since it's a new package they say it could take a long time before they get to reviewing it.  I don't know how long that is.