AlexandreRouma/SDRPlusPlus #1462: Client crashes on packet loss while connected to Server

lalliexperience, 2024-09-01:
** Hardware **
Multiple computers, Mac or PC versions as client, using Pi or Linux PC as SDRPP server. The symptoms are identical on any hardware for client and server.

**Software**
Multiple operating systems, Mac and PC versions as client, using Pi Linux PC as SDRPP Server.  The symptoms are identical on any hardware for client and server.

**Bug Description**
The SDR++ Client Software will freeze/crash any time there is a network interruption while connected to a SDRPP Server.
It would be nice if the software could detect this and just disconnect instead of freezing.

The SDR++ Client Software will freeze when you attempt to connect to  a SDRPP Server and port which does not exist.
It should know fairly quickly that there is not a service on that destination port and provide an error.

**Steps To Reproduce**
- Unplug ethernet/Unplug router/Disconnect wifi/ cause packet loss in any way between the SDRPP Client and SDRPP Server. Client software will freeze permanently. Can not click Disconnect. Must force close software (Windows or Mac)

- Attempt to connect to an SDRPP Server IP and port that does not exist.

**Screenshots**
When Packet loss: cant click disconnect, permanent freeze, must force quit program:
[image]

Attempting to connect to a non existing IP and port, 30 second freeze:
[image]

**Additional info**
The Software running in Server mode does not seem to be effected by interuptions.

AlexandreRouma, 2024-09-01:
Are you sure the freeze is permanent? The timeout on connection is set by the operating system and the timeout on commands is set to 10 seconds. On my PC the connecting to a non-server IP freeze (which is expected since everything is synchronous in the UI) only lasts 20 seconds, and I can't test the other situation atm.

lalliexperience, 2024-09-01:
For attempting connection to non existent Ip/port it’s not permanent, but
it is a while.  I just retested on Mac and it was froze for over 1 minute.

For loss of connection it’s a complete freeze that never recovers.