AlexandreRouma/SDRPlusPlus #1463: SoapySDR module missing

daochild, 2024-09-03:
**Hardware**
- CPU: 4Cores
- RAM: 6GB
- GPU: N/A
- SDR: SoapySDR

**Software**
- Operating System: De
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**
SoapySDR not found in the modules list

**Steps To Reproduce**
SoapySDR driver installed
SDR++ Installed by instruction on github
No errors!

[image]

daochild, 2024-09-03:
Do I need pass any parameters in cmake for this SDR?

AlexandreRouma, 2024-09-03:
https://github.com/AlexandreRouma/SDRPlusPlus/issues/1455#issuecomment-2306309845

closed 2024-09-03 by AlexandreRouma