AlexandreRouma/SDRPlusPlus #1470: Panadapter manager

Flo250578, 2024-09-11:
hello  I come up with the idea of ​​implementing a standalone panadapter module on an android device.  This type of panadapter does not need to be synchronized with the transceiver.  it only monitors the spectrum around the intermediate frequency.  having a fixed frequency on the sdr, I need to be able to block this frequency so as not to change it accidentally.  for example, with my ft-891 I only need the frequency 69.450mhz.  practically once blocked it becomes zero fundamental frequency.  and what is left and right can be measured in + or - kHz.  my true frequency is on the transceiver display.  so on the sdr application it can become invisible.  if you find it interesting, we can develop it with more technical data.
[image]
[image]

AlexandreRouma, 2024-09-11:
If it doesn't need to follow the frequency, then just set the frequency in center tuning mode and don't touch it.
If it needs to follow the frequency, there's already the rigctl client module.

closed 2024-09-11 by AlexandreRouma

Flo250578, 2024-09-11:
Smart thing.  I didn't even think about that. (sarcasm)

AlexandreRouma, 2024-09-11:
You're asking me to create an entire module just to lock the frequency which is a feature that has already been requested. And you know that very well since you just commented on it.
I don't need your attitude, fuck off.