AlexandreRouma/SDRPlusPlus #1471: SDR++ Nightly Build (currently 1.2.0) doesn't have support for RTL-SDR on .deb files for Linux

mrgadotti, 2024-09-11:
**Software**
- Operating System: Ubuntu 24.04
- SDR++: Nightly Build (currently 1.2.0)

**Bug Description**
- SDR++ Nightly Build (currently 1.2.0) doesn't have support for RTL-SDR on .deb files for Linux
- Before this download from Nightly, it's working fine on older version instaled before 2024-09-10
- Same error on Linux Mint 22
- I did build from sources and the RTL-SDR support return and it's working again
- MacOs build is working fine with RTL-SDR support

**Steps To Reproduce**
1. Download SDR++ Nightly Build (currently 1.2.0) for Ubuntu (sdrpp_ubuntu_noble_amd64.deb)
2. Install
3. Run

**Screenshots**

[image]

AlexandreRouma, 2024-09-11:
Works fine here, make sure you're up to date.

closed 2024-09-11 by AlexandreRouma

mrgadotti, 2024-09-11:
I did try on another VM from Linux Mint 21.2, with clean install, same error. The RTL-SDR doesn't appear on the list...
Interesting that it only occurs on Linux, on MacOS it works fine, list the RTL-SDR

[image]

mrgadotti, 2024-09-11:
Build from yesterday, 2024-09-10

AlexandreRouma, 2024-09-11:
You're probably using the wrong deb.

AlexandreRouma, 2024-09-11:
is librtlsdr installed on that clean install?

mrgadotti, 2024-09-11:
I need to really apologize, it works after install librtlsdr. Works on Ubuntu and Mint

sudo apt install librtlsdr2 librtlsdr-dev