AlexandreRouma/SDRPlusPlus #1472: Sound Network UDP

gr8lite, 2024-09-11:
*x86_64
RTL_SDR
- Operating System: Ubuntu 24.04
- SDR++: Version 1.2.0 - Sept. 10th 2024

**Bug Description**
RADIO - selecting NETWORK causes SDRPP to immediately close.

It had been working in the past. 

I have reinstalled many times using various methods, but con not set to NETWORK/UDP

I had a problem with starting the SDRPP , but resolved it following instructions to remove soapy .json reference file.

SDRPP functions normally other than RADIO/NETWORK configuration.

**Steps To Reproduce**
1. ...click drop down under "RADIO"
2. ...  Select "NETWORK"
3. ...  Watch SDRPP close

[image]

AlexandreRouma, 2024-09-11:
Can't replicate, show SDR++ logs.

gr8lite, 2024-09-11:
I am not sure what you would like.   
What can I do to help?

gr8lite, 2024-09-11:
the following it the output when I sdrpp -c

:/$ sdrpp -c
[11/09/2024 11:36:34.000] [INFO] SDR++ v1.2.0
Unknown argument
-a --addr               Server mode address
   --autostart          Automatically start the SDR after loading
-h --help               Show help
-p --port               Server mode port
-r --root               Root directory, where all config files are stored
-s --server             Run in server mode

AlexandreRouma, 2024-09-11:
on linux just run it without argument.

gr8lite, 2024-09-11:
When I command line "sdrpp"  It opens the app normally.   
However if I go to the menu and under RADIO I select NETWORK - The sdrpp shuts down immediately.~$ sdrpp
[11/09/2024 12:45:24.000] [INFO] SDR++ v1.2.0
[11/09/2024 12:45:24.000] [INFO] Loading config
[11/09/2024 12:45:24.000] [WARN] ConfigManager locked, waiting...
[11/09/2024 12:45:24.000] [INFO] Using OpenGL 3.0
[11/09/2024 12:45:24.000] [INFO] Loading icons
[11/09/2024 12:45:24.000] [INFO] Loading band plans
[11/09/2024 12:45:24.000] [INFO] Loading band plans color table
[11/09/2024 12:45:24.000] [INFO] Loading modules
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/pager_decoder.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/perseus_source.so
[11/09/2024 12:45:24.000] [WARN] ConfigManager locked, waiting...
[11/09/2024 12:45:24.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/perseus_source.so: libperseus-sdr.so.0: cannot open shared object file: No such file or directory
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfnm_source.so
[11/09/2024 12:45:24.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/rfnm_source.so: librfnm.so: cannot open shared object file: No such file or directory
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/radio.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/file_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/m17_decoder.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_client.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/recorder.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_sink.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/iq_exporter.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/scanner.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_source.so
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[11/09/2024 12:45:24.000] [ERROR] Couldn't load /usr/lib/sdrpp/plugins/limesdr_source.so: libLimeSuite.so.20.10-1: cannot open shared object file: No such file or directory
[11/09/2024 12:45:24.000] [INFO] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[11/09/2024 12:45:24.000] [INFO] Initializing Audio Sink (audio_sink)
[11/09/2024 12:45:24.000] [INFO] Initializing Audio Source (audio_source)
[11/09/2024 12:45:24.000] [INFO] Initializing File Source (file_source)
[11/09/2024 12:45:24.000] [INFO] Initializing Frequency Manager (frequency_manager)
[11/09/2024 12:45:24.000] [INFO] Initializing Network Sink (network_sink)
[11/09/2024 12:45:24.000] [INFO] Initializing RTL-SDR Source (rtl_sdr_source)
[11/09/2024 12:45:24.000] [INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[11/09/2024 12:45:24.000] [INFO] Initializing Radio (radio)
[Resamp] predec: 32, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[Resamp] predec: 128, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 1, interp: 24, decim: 25, inacc: 0.000000%, taps: 1900
[Resamp] predec: 1, interp: 24, decim: 25, inacc: 0.000000%, taps: 1900
[11/09/2024 12:45:24.000] [WARN] Demod switch took 1172 us
[11/09/2024 12:45:24.000] [INFO] RtAudio stream open
[11/09/2024 12:45:24.000] [INFO] Initializing Recorder (recorder)
[11/09/2024 12:45:24.000] [INFO] Initializing Rigctl Server (rigctl_server)
[11/09/2024 12:45:24.000] [INFO] Initializing SDR++ Server Source (sdrpp_server_source)
[11/09/2024 12:45:24.000] [INFO] Initializing SDRplay Source (sdrplay_source)
[11/09/2024 12:45:24.000] [INFO] Initializing SpyServer Source (spyserver_source)
[11/09/2024 12:45:24.000] [INFO] Loading color maps
[... cut ...]
[11/09/2024 12:45:25.000] [INFO] Running post-init for SDRplay Source
[11/09/2024 12:45:25.000] [INFO] Running post-init for SpyServer Source
[11/09/2024 12:45:25.000] [INFO] Ready.
[11/09/2024 12:45:30.000] [INFO] Exiting successfully
 ******************************************************************************************
AFTER RADIO / NETWORK
[Resamp] predec: 1, interp: 24, decim: 25, inacc: 0.000000%, taps: 1900
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could get address from host
Aborted (core dumped)

gr8lite, 2024-09-11:
[Resamp] predec: 1, interp: 24, decim: 25, inacc: 0.000000%, taps: 1900
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could get address from host
Aborted (core dumped)

gr8lite, 2024-09-11:
[image]

AlexandreRouma, 2024-09-11:
Seems the last time you used the network sink, you entered an invalid host name in the host field. I added a check for this, install the latest nightly and check if it still crashes.

gr8lite, 2024-09-11:
Although I had not notice in the past, I think a vpn was interfering with it.  
I was trying to use UDP locally so I didn't give it any thought.
With vpn off the NETWORK /UDP does appear to work.
I guess that is good news.
Now the question is - can I use the NETWORK/UDP  locally while using vpn for web browser at the same time?

gr8lite, 2024-09-11:
will do - thank you.

(Seems the last time you used the network sink, you entered an invalid host name in the host field. I added a check for this, install the latest nightly and check if it still crashes.(

gr8lite, 2024-09-12:
seems to be fine now.

closed 2024-09-12 by AlexandreRouma