AlexandreRouma/SDRPlusPlus #1474: SDR++ jammy file not opening on Ubuntu 22.04...

TyFromDeep, 2024-09-13:
Hello, let me start off by saying I am brand new to pi's and ubuntu. I am having a problem with opening SDR++.  When I installed the jammy.deb file from nightly builds the app was finally showing in applications, but it does nothing when I try to open it from there. So, I went to terminal to try and open it using sdrpp -c. I got this message "x86_64-binfmt-P: Could not open '/lib64/ld-linux-x86-64.so.2': No such file or directory". I am on a pi400 with ubuntu 22.04. I've gone through many discussion threads with similar problems but none of their fixes are working for me.

AlexandreRouma, 2024-09-13:
the file you downloaded has `amd64` in the name, why would you expect it to run on an ARM CPU?

TyFromDeep, 2024-09-13:
well, I am brand new to this, my first time ever using a pi... 

so, if that jammy file wont work for me i'm guessing I just can't have sdr++ on this device?

AlexandreRouma, 2024-09-13:
No, you have to build from source.