AlexandreRouma/SDRPlusPlus #1480: Expand Audio output devices menu

c3kkos, 2024-09-25:
**Feature description**

It happens to have more then 3 output devices (sinks) in modern, integrated audio adapters.
I have a HP ultrabook that, via pipewire, offers many endpoints (3 hdmi and 1 duplex analog output)

sdrpp only show 3 out of 4, all hdmi, making it impossible to choose loudspeakers/headphones combo output.

Please expand to show as many endpoints found

[image]

Workarounds:

manually edit xml in ~/.config 

use qpwgraph to manually re-route audio output to the correct sink

AlexandreRouma, 2024-09-25:
All available devices are always shown, it doesn't limit anything. Show logs.

c3kkos, 2024-09-26:
Sure.

```
sdrpp
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] SDR++ v1.2.0
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading config
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Using OpenGL 3.0
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading icons
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading band plans
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading band plans color table
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading modules
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/radio.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/pager_decoder.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/tetra_demodulator.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/rigctl_client.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/iq_exporter.so
[26/09/2024 21:20:50.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/scanner.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/network_source.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/recorder.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/file_source.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/network_sink.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/audio_source.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Initializing Audio Sink (audio_sink)
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Initializing File Source (file_source)
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Initializing Frequency Manager (frequency_manager)
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Initializing RTL-SDR Source (rtl_sdr_source)
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Initializing RTL-TCP Source (rtl_tcp_source)
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Initializing Radio (radio)
[Resamp] predec: 32, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

[Resamp] predec: 128, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 1, interp: 24, decim: 25, inacc: 0.000000%, taps: 1900
[Resamp] predec: 1, interp: 24, decim: 25, inacc: 0.000000%, taps: 1900
[26/09/2024 21:20:51.000] [<font color="#C4A000">WARN</font>] Demod switch took 879 us
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] RtAudio stream open
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Initializing Recorder (recorder)
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Initializing SDRplay Source (sdrplay_source)
shm_open: No such file or directory
[26/09/2024 21:20:51.000] [<font color="#CC0000">ERROR</font>] Could not intiatialized the SDRplay API. Make sure that the service is running.
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Initializing SpyServer Source (spyserver_source)
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Loading color maps
[... cut ...]
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Running post-init for Recorder
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Running post-init for SDRplay Source
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Running post-init for SpyServer Source
[26/09/2024 21:20:51.000] [<font color="#4E9A06">INFO</font>] Ready.
```

Indeed, there is a jack related error, from its api, but 1) not sure if that affects the issue 2) why jack? anyhow, i've got pipewire and wireplumber, up&running fine, with alsa and jack support enabled (Cockos Reaper works)

AlexandreRouma, 2024-09-26:
Because as I suspected it's a rtaudio issue and not a SDR++ one. Follow [these instructions](https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112) for a fix.

c3kkos, 2024-09-27:
Thank you Alexander! 

Just in case someone wants a quick&dirty workaround for this:

in Linux Mint latest releases Pipewire and Wireplumber audio systems are the new defaults, so you should have only to install "qpwgraph"

that lets you play with nodes.

Just detach the current "Radio" (sdrpp) and attach the channels to your physical output device. 

[image]

c3kkos, 2024-09-27:
I've done what written for rtaudio. Followed every step, including checking out v5.1

but got a very ugly seg fault when i start sdrpp.

here's the log:

```
sdrpp 
[27/09/2024 22:25:59.000] [INFO] SDR++ v1.2.0
[27/09/2024 22:25:59.000] [INFO] Loading config
[27/09/2024 22:26:00.000] [INFO] Using OpenGL 3.0
[27/09/2024 22:26:00.000] [INFO] Loading icons
[27/09/2024 22:26:00.000] [INFO] Loading band plans
[27/09/2024 22:26:00.000] [INFO] Loading band plans color table
[27/09/2024 22:26:00.000] [INFO] Loading modules
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/radio.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/pager_decoder.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/tetra_demodulator.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_client.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/iq_exporter.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/scanner.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/recorder.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/file_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/network_sink.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/audio_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[27/09/2024 22:26:00.000] [INFO] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[27/09/2024 22:26:00.000] [INFO] Initializing Audio Sink (audio_sink)
[27/09/2024 22:26:00.000] [INFO] Initializing File Source (file_source)
[27/09/2024 22:26:00.000] [INFO] Initializing Frequency Manager (frequency_manager)
[27/09/2024 22:26:00.000] [INFO] Initializing RTL-SDR Source (rtl_sdr_source)
[27/09/2024 22:26:00.000] [INFO] Initializing RTL-TCP Source (rtl_tcp_source)
[27/09/2024 22:26:00.000] [INFO] Initializing Radio (radio)
[Resamp] predec: 32, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500

RtApiDummy: This class provides no functionality.

[1]    9863 segmentation fault (core dumped)  sdrpp

```

AlexandreRouma, 2024-09-27:
It wasn't followed properly or you didn't have at least pulseaudio or pipewire-pulse instead.

c3kkos, 2024-09-27:
I tried to build and install fresh but none worked.

i do have pipewire-pulse

```
sudo apt install pipewire-pulse
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
pipewire-pulse is already the newest version (1.0.7-3~ubuntu24.04).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

```

sudo make uninstall <- in rtaudio src dir and reverting to packaged dev lib fixed the segmentation fault

AlexandreRouma, 2024-09-27:
Can't help any more than that. it's clearly an rtaudio issue on your specific system so I have no control over it. Open up an issue with them and they might be able to help you.

closed 2024-09-27 by AlexandreRouma

c3kkos, 2024-09-27:
2nd time i tried it worked... i don't know, probably i messed up something..

[27/09/2024 22:52:52.000] [INFO] RtAudio stream open

sorry for that. i do confirm now is all working