AlexandreRouma/SDRPlusPlus #1481: Can’t find recording made on Android

trent78y, 2024-09-27:
I made a recording recently and I cannot find it anywhere in the files app. I made the recording on a Samsung A51 and when I go into settings I can see the app is taking up 900MB so there is something saved but I don’t know where.

c3kkos, 2024-09-28:
try listing all files in memory and sorting for decreasing size?

trent78y, 2024-09-28:
> try listing all files in memory and sorting for decreasing size?

Sorry, I have tried lots of different file managers, adb. Not really sure where to go next.

I should have probably clarified in the original post that I am NOT rooted.

AlexandreRouma, 2024-09-28:
the default path is in the app's own directory and modern android doesn't let you access it. You need ADB to get there.

Next time change the recording path to your SD card or something

trent78y, 2024-09-28:
> the default path is in the app's own directory and modern android doesn't let you access it. You need ADB to get there.
> 
> Next time change the recording path to your SD card or something

I have tried adb and looked in so many directories but I still can’t find it.

AlexandreRouma, 2024-10-03:
You need to go into the app's private directory.

closed 2024-10-03 by AlexandreRouma

dsaderholm, 2024-10-04:
I am using adb to access this directory. I can see the two recordings there. Now how can I download them or access them?

trent78y, 2024-10-04:
> I figured it out. This command will put those files into whatever direcory you ran ADB in:
> 
> `adb exec-out run-as org.sdrpp.sdrpp tar -cf - -C /data/user/0/org.sdrpp.sdrpp/files/recordings . | tar xf -`

This command saved my recording!!!!! Thank you so much!!!!!