AlexandreRouma/SDRPlusPlus #1489: Please update to the latest libiio release, version 0.26.

simon830130, 2024-10-08:
PlutoSDR and similar products require the latest version of libiio to search for the device online on the Windows platform. The straightforward method is to download the file named Windows.zip from the libiio releases page, extract the libiio.dll from the zip file, and replace the original dll that came with SDRPP. However, having the application built-in would be ideal.

ashjmckenzie, 2024-11-17:
I found simply overwrighting libiio.dll removes the PlutoSDR option all together in the lastest nightlys: I had to grab more:

Download https://github.com/analogdevicesinc/libiio/releases/download/v0.26/Windows.zip

From: Windows.zip\Windows-VS-2022-x64
Copy the following to the lastest build:
- libserialport-0.dll
- libiio-sharp.dll
- libiio.dll (this will overight)
- libxml2.dll

Then SDR++ will autodetect PlutoIP Devices

povnz, 2025-03-23:
Ok recently to make this work I had to disable IPv6 as there seems to be a conflict if the Pluto is available on both IPv4&6