AlexandreRouma/SDRPlusPlus #1492: Interface suggestion

ewhansen, 2024-10-12:
Will you consider an option:  When the mouse cursor is hovered over the spectrum or waterfall, the frequency is displayed?  I want to know the frequency of a peak without actually clicking on it and changing where the radio is tuned.  Thanks.

AlexandreRouma, 2024-10-12:
Someone could make a plugin for it but I don't like having stuff show up next to my cursor while hovering the waterfall or FFT so it's a no from me.

closed 2024-10-12 by AlexandreRouma

ewhansen, 2024-10-13:
Agreed that having it on all the time would be annoying.  I had in mind that it could be an option turned on or off, as needed, perhaps with option- or alt- key.  It wouldn't be for tracking a peak, just getting a quick estimate of frequency that is better than eyeballing the horizontal axis.