AlexandreRouma/SDRPlusPlus #1494: Paste frequency via Ctrl+V

larsen0815, 2024-10-13:
**Feature description**
I have a frequency that I copied from a webpage. Would be nice to just Ctrl+V to set the frequency in SDR++

AlexandreRouma, 2024-10-15:
good idea, I'll also make CTRL+C copy the current frequency

closed 2024-10-16 by AlexandreRouma

michaelgeorge001, 2024-10-16:
Looks good except for a few exceptions, at least on my Mac running Ventura 13.7. Seems to break on copying decimal frequencies like 10.0MHz, 20.0MHz, 30.0MHz ...  100MHz, 200MHz, 300MHz et al. Pasting the copy results in 1MHz, 2MHz, 3MHz ...1MHz, 2MHz, 3MHz respectively.  This also appears in the KHz and GHZ ranges. Seems to be related to trimming the zeroes.

AlexandreRouma, 2024-10-16:
Copy pasting these exact strings from your comment on windows and linux works as expected so I don't know why it wouldn't work on MacOS. I don't have a working mac so I can't test anything.

michaelgeorge001, 2024-10-16:
Not a problem. I'll try different HW/SW configs on my end and see what additional details I can provide. It's definitely not a showstopper.

jeffrizzo, 2024-10-16:
I'm guessing the latest nightly hasn't built this yet, as CMD-C doesn't copy anything for me on macos (just installed).  I can try again after it's built again

AlexandreRouma, 2024-10-16:
the latest nightly has it

AlexandreRouma, 2024-10-16:
@michaelgeorge001 Actually I misunderstood what you were describing, I can replicate it now, fixing in a sec

michaelgeorge001, 2024-10-16:
Not a problem. I was in the process of appending my comment, once I realized the source<->destination was not clearly indicated.

AlexandreRouma, 2024-10-16:
Should be fixed but it'll take 15 minutes or so for a new nightly to be done building.

michaelgeorge001, 2024-10-16:
Works perfectly. Thank you.

larsen0815, 2024-10-16:
Thanks for the quick implementation 👍