AlexandreRouma/SDRPlusPlus #1498: SDRPlusPlus does not support rtaudio 6.0.1

hobbes1069, 2024-10-20:
**Software**
- Operating System: Fedora Rawhide
- SDR++: 1.2.0 commit 052167962dbf9adc2a02825f2f428e7613255d50

**Bug Description**
When building with rtaudio 6.0.1 the following errors occur:
```
/builddir/build/BUILD/sdrpp-1.2.0-build/SDRPlusPlus-052167962dbf9adc2a02825f2f428e7613255d50/source_modules/audio_source/src/main.cpp: In member function ‘void AudioSourceModule::refresh()’:
/builddir/build/BUILD/sdrpp-1.2.0-build/SDRPlusPlus-052167962dbf9adc2a02825f2f428e7613255d50/source_modules/audio_source/src/main.cpp:93:26: error: ‘struct RtAudio::DeviceInfo’ has no member named ‘probed’
   93 |                 if (info.probed && info.inputChannels < 2) { continue; }
      |                          ^~~~~~        

/builddir/build/BUILD/sdrpp-1.2.0-build/SDRPlusPlus-052167962dbf9adc2a02825f2f428e7613255d50/sink_modules/audio_sink/src/main.cpp:175:16: error: ‘RtAudioError’ does not name a type; did you mean ‘RtAudioErrorType’?
  175 |         catch (RtAudioError& e) {
      |                ^~~~~~~~~~~~
      |                RtAudioErrorType
/builddir/build/BUILD/sdrpp-1.2.0-build/SDRPlusPlus-052167962dbf9adc2a02825f2f428e7613255d50/sink_modules/audio_sink/src/main.cpp:175:16: error: ‘RtAudioError’ does not name a type; did you mean ‘RtAudioErrorType’?
  175 |         catch (RtAudioError& e) {
      |                ^~~~~~~~~~~~
      |                RtAudioErrorType
```

AlexandreRouma, 2024-10-20:
that is a 10 month old commit, what are you doing?

AlexandreRouma, 2024-10-20:
I bet you didn't read this
[image]

closed 2024-10-20 by AlexandreRouma