AlexandreRouma/SDRPlusPlus #1502: uInt8 baseband recording appears to play back at twice the speed

mpentler, 2024-10-22:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel Core i7
- RAM: 16 GB
- GPU: RTX 4070
- SDR: Local, HackRF

**Software**
- Operating System: Name + Exact version: Windows 11 x64 (up to date)
- SDR++: Version + Build date: 1.2.0 oct 22nd 2024 nightly

**Bug Description**
I have recorded a 10 MHz chunk of the FM broadcast spectrum with a HackRF in uint8 format to save space. Upon loading this file back into SDR++ and hitting play the file plays back at what I perceive to be twice the speed.

**Steps To Reproduce**
1. Select radio and tune a centre frequency with a selected sample rate
2. Change recording format to baseband, uInt8
3. Record some minutes of baseband
4. Load this baseband in using the File source and hit play

**Additional info**
I have not tested any other sample rates or SDRs as I need to do something very specific tonight so I have just gone with Int16 for now. I can test other things on request and have an RSP1a, an RSP1 clone, and an RTL-SDR stick available.

AlexandreRouma, 2024-10-22:
not a bug, the file player simply only supports int16 and float32. I'll change this to a feature request for supporting other bit depths.

mpentler, 2024-10-22:
Copy that!

qrp73, 2025-03-30:
to save space you can record it to the flac container, in my mod it is already implemented: https://github.com/qrp73/SDRPP
If you want to record audio, you can also select MP3 container for recording it allows to write 10 times smaller files.

It also supports to play WAV files in PCM 8/16/24/32 bit or float32 sample format. 

[image]