AlexandreRouma/SDRPlusPlus #1508: Sometimes toggling the FFT smoothing toggle crashes the program

KubaPro010, 2024-11-02:
**Hardware**
- CPU: Intel Core i5-9400F
- RAM: 16 GB
- GPU: GTX 1070
- SDR: Remote MiriSDR

**Software**
- Operating System: Windows 11 x64
- SDR++: Oct 14 2024 (self-compiled, but i didn't see commits regarding this)

**Bug Description**
Toggling the FFT smoothing switch makes SDR++ hang (probably indefinitely)

**Steps To Reproduce**
1. Maybe wait some time
2.  Toggle FFT smoothing toggle

AlexandreRouma, 2024-11-06:
can't replicate at all, you're gonna have to be more detailed

KubaPro010, 2024-11-06:
what do you mean by "more detailed", i can also tell you that it mostly happens when receiving is enabled

AlexandreRouma, 2024-11-06:
Screen recording of the issue, state of other settings, etc

KubaPro010, 2024-11-06:
managed to crash it toggling the switch repeatedly, will send the video in a moment (in the video, the aftermath of this is that sdr++ will keep not responding until you close it, console logs also don't see anything as last log is `[06/11/2024 22:01:04.000] [INFO] SDRPPServerSourceModule 'SDR++ Server Source': Tune: 95500000.000000!`)

AlexandreRouma, 2024-11-06:
Since you're building yourself, try to get a stack trace

KubaPro010, 2024-11-06:
actually this might have been a invalid issue, I have a oct 7 build and it did the same thing but the latest build doesn't crash by the same thing

AlexandreRouma, 2024-11-06:
there was no change to any of that part of the software

AlexandreRouma, 2025-02-21:
Closing for inactivity since no backtrace was posted.

closed 2025-02-21 by AlexandreRouma