AlexandreRouma/SDRPlusPlus #1509: Compilation fails with clang-19: implicit instantiation of undefined template 'std::char_traits<fmt::internal::char8_type>'

yurivict, 2024-11-03:
# WARNING: Filling out the template below is NOT optional. Issues not filling out this template will be closed without review.

FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
Irrelevant: compilation issue.

**Software**
- Operating System: FreeBSD 14.1
- SDR++: 1.0.4

**Bug Description**
```
In file included from /wrkdirs/usr/ports/comms/sdr++/work/SDRPlusPlus-1.0.4/src/main.cpp:1:
In file included from /wrkdirs/usr/ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/core.h:3:
In file included from /wrkdirs/usr/ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/module.h:5:
In file included from /wrkdirs/usr/ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/utils/event.h:3:
In file included from /wrkdirs/usr/ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/spdlog/spdlog.h:12:
In file included from /wrkdirs/usr/ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/spdlog/common.h:36:
In file included from /wrkdirs/usr/ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/spdlog/fmt/fmt.h:20:
/wrkdirs/usr/ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/spdlog/fmt/bundled/core.h:378:30: error: implicit instantiation of undefined template 'std::char_traits<fmt::internal::char8_type>'
  378 |       : data_(s), size_(std::char_traits<Char>::length(s)) {}
      |                              ^
/wrkdirs/usr/ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/spdlog/fmt/bundled/format.h:580:9: note: in instantiation of member function 'fmt::basic_string_view<fmt::internal::char8_type>::basic_string_view' requested here
  580 |       : basic_string_view<internal::char8_type>(
      |         ^
/usr/include/c++/v1/__fwd/string.h:23:29: note: template is declared here
   23 | struct _LIBCPP_TEMPLATE_VIS char_traits;
      |                             ^
1 error generated.
```

**Steps To Reproduce**
Regular build

AlexandreRouma, 2024-11-03:
You're seriously reporting a "bug" for a 3 year old version?

closed 2024-11-03 by AlexandreRouma

yurivict, 2024-11-03:
> You're seriously reporting a "bug" for a 3 year old version?

Did a miss a newer version? Where is it?

AlexandreRouma, 2024-11-03:
SDR++ moved to a rolling release model 3 years ago. There are commits to the master branch on average almost every day.
https://github.com/AlexandreRouma/SDRPlusPlus/releases/tag/nightly

yurivict, 2024-11-03:
I didn't know that, thanks.

yurivict, 2024-11-03:
@AlexandreRouma 
FYI: All other packagers aren't aware of a rolling release either: https://repology.org/project/sdr%2B%2B/versions
The reason is that they go by GitHub tags which are automatically discovered by release auto-discovery systems.

Users have outdated , frozen versions, and these days users expect to have pre-built packages for everything.

You might want to do one of these things:
1. Reach out to maintainers of these packages.
2. Add tags like YYYYMMDD periodically, like once a month.